dchristl/macless-haystack #4: Get FindMy report in Python to get my position in AppDaemon

CrazYoshi, 2023-03-07:
Hi,
first of all, thanks for sharing your code. It looks very promising and for developers like me that are not using MacOS is really useful. 
I would like to implement an easy script in python to get my openhaystack device position and integrate it in my AppDaemon and HomeAssistant.
Could you please give me some hints in how to do that? How can I get the reports from the FindMy network to achieve that?

Thanks you!

dchristl, 2023-03-12:
Hello CrazyYoshi,

you could have a look at [Biemster's Find my](https://github.com/biemster/FindMy). There is a client and a webserver and the base of the webserver part of my project (s. webserver-folder). Sadly it is not possible to use it all without any OSX-machine (real or virtual), because part of the authetication can only obtained by the MacOS-keychain and there it is generated in (currently) non reversed engineered ways. The objc - dependency is used for that, so it is needed to get part of the webserver call.

Regards
Danny

mrmay-dev, 2023-04-21:
Home Assistant / OpenHab / HomeBridge integration can be done with OwnTracks and MQTT. The trick is to format the OpenHaystack reports as OwnTracks location updates. I track my Haystack devices like that. Best of all is my son uses Android and I'm on iOS and we can track shared items together. Here's something I wrote about [how it can be done](https://github.com/seemoo-lab/openhaystack/discussions/195).

Informatic, 2024-02-09:
In case anyone stumbles upon this issue - there's a python implementation of crypto algorithm used by airtags and Influxdb ingesting daemon (*should* work with mh_endpoint service from macless-haystack too):
https://github.com/hatomist/openhaystack-python
https://github.com/hatomist/openhaystack-grafana

Here's a very simple python PoC that works with mh_endpoint here: https://gist.github.com/Informatic/87bab2f8e5466e0e3fd5965a458b0795 (`airtagcrypto` package is above `openhaystack-python`)

closed 2024-04-22 by dchristl