dchristl/macless-haystack #6: Webserver Python errors on padding

aircable, 2023-04-11:
With Python 3.11 and the newest cryptography package I get this error:
```
  File "headless-haystack/webserver/request_reports.py", line 142, in <module>
    iCloud_decryptionkey = args.key if args.key else retrieveICloudKey()
                                                     ^^^^^^^^^^^^^^^^^^^
  File "headless-haystack/webserver/request_reports.py", line 104, in retrieveICloudKey
    db_key = unpad(decrypt(db_key_enc, algorithms.TripleDES(master_key), modes.CBC(db_key_IV)), algorithms.TripleDES.block_size)[:24]
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "headless-haystack/webserver/request_reports.py", line 36, in unpad
    return unpadder.update(paddedBinary) + unpadder.finalize()
                                           ^^^^^^^^^^^^^^^^^^^
  File "python3.11/site-packages/cryptography/hazmat/primitives/padding.py", line 159, in finalize
    result = _byte_unpadding_check(
             ^^^^^^^^^^^^^^^^^^^^^^
  File "python3.11/site-packages/cryptography/hazmat/primitives/padding.py", line 101, in _byte_unpadding_check
    raise ValueError("Invalid padding bytes.")
ValueError: Invalid padding bytes.
```

dchristl, 2023-04-12:
Hello aircable,

I tested it out and it works for me with the newest version. Do you use it with the docker image or with a real mac? Have you checked the password at the top of the file (line 8) and changed it, if it is not 'alpine'?

aircable, 2023-04-12:
I run on a real mac with Python installed from brew.
The error is a Python execution error. There are actually more errors
showing that the code was written for an older version of Python.
FindMy_proxy.py line125
bytes needs parameter for encoding:

                            iterations=1000,
backend=default_backend()).derive(bytes(pwd, encoding='utf8'))

aircable, 2023-04-14:
> Yes, you are right, I forget to change a piece of code before publish.
> I've changed this now (s. latest release
> and it should work on python 2 and 3 now

Thanks for the changes. But I'm still getting this error.
Must have something to do with the cryptography library.
I just PIP installed it with the latest version.

Using python3

Keychain password: alpine

Traceback (most recent call last):

  File "headless-haystack/webserver/FindMy_proxy.py", line 75, in <module>

    retrieveICloudKey()

  File "headless-haystack/webserver/apple_cryptography.py", line 51, in
retrieveICloudKey

    db_key = unpad(decrypt(db_key_enc, algorithms.TripleDES(master_key),
modes.CBC(db_key_IV)), algorithms.TripleDES.block_size)[:24]

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "headless-haystack/webserver/apple_cryptography.py", line 36, in
unpad

    return unpadder.update(paddedBinary) + unpadder.finalize()

                                           ^^^^^^^^^^^^^^^^^^^

  File
"/usr/local/lib/python3.11/site-packages/cryptography/hazmat/primitives/padding.py",
line 159, in finalize

    result = _byte_unpadding_check(

             ^^^^^^^^^^^^^^^^^^^^^^

  File
"/usr/local/lib/python3.11/site-packages/cryptography/hazmat/primitives/padding.py",
line 101, in _byte_unpadding_check

    raise ValueError("Invalid padding bytes.")

ValueError: Invalid padding bytes.

aircable, 2023-04-14:
never mind, you can close that thread
my bad

closed 2023-04-15 by dchristl