dchristl/macless-haystack #7: This application requires macOS11 or higher

SimonBard, 2023-04-21:
I started the container via:

```
docker run -it \
    --device /dev/kvm \
    -p 50922:10022 \
    -p 56176:6176 \
    -p 56443:6443 \
    -v /tmp/.X11-unix:/tmp/.X11-unix \
    -e "DISPLAY=${DISPLAY:-:0.0}" \
    -e GENERATE_UNIQUE=true \
    -e ADDITIONAL_PORTS='hostfwd=tcp::6176-:6176,hostfwd=tcp::6443-:6443,' \
    -e WIDTH=1024 \
    -e HEIGHT=768 \
    -e MASTER_PLIST_URL=https://raw.githubusercontent.com/dchristl/headless-haystack/main/OSX-KVM/custom.plist \
    sickcodes/docker-osx:auto
```

In the container, I downloaded OpenHaystack and started it. Got the error message: "You can't use this version fo the application "OpenHAystack " with this version of macOS."

I then tried:

```
docker run -it \
    --device /dev/kvm \
    -p 50922:10022 \
    -p 56176:6176 \
    -p 56443:6443 \
    -v /tmp/.X11-unix:/tmp/.X11-unix \
    -e "DISPLAY=${DISPLAY:-:0.0}" \
    -e GENERATE_UNIQUE=true \
    -e ADDITIONAL_PORTS='hostfwd=tcp::6176-:6176,hostfwd=tcp::6443-:6443,' \
    -e WIDTH=1024 \
    -e HEIGHT=768 \
    -e MASTER_PLIST_URL=https://raw.githubusercontent.com/dchristl/headless-haystack/main/OSX-KVM/custom.plist \
    sickcodes/docker-osx:big-sur
```

but it would not startup, instead I had to install the system, but failed as it said "disk locked" in some step.

dchristl, 2023-04-24:
Hello @SimonBard ,

have you read the [instructions](https://github.com/dchristl/headless-haystack/blob/main/OSX-KVM/README.md). The whole purpose of this project is to avoid the installation of OpenHaystack.

backupandforkacc, 2023-04-28:
Ok got it. 

I can reach the server with the given port number. 
I do not get any reports. I am using an ESP32 and have used esptool to write the data on the esp. I have carried the esp32 with and without an apple iphone SE 2000 (which uses the same apple account as the headless server). 

Can I also track original apple airpods? This would faciliate debugging a bit.

dchristl, 2023-04-28:
Hard to tell what is gone wrong. I think there is something wrong with the firmware. Can you post a screenshot of the Espressif UI? Are you able to get a log out of your esp32? The firmware logs everytime when a key will be advertised. 

Sadly, it is not possible to track original Apple devices, because you don't have the private key for decryption.

backupandforkacc, 2023-04-29:
[image]

Many thanks for the feedback!

I have never worked with espressif, but did now. 
How can I monitor logs? I have used Arduino IDE for seeing the logs or sent them by mqtt.

dchristl, 2023-04-29:
First of all, I see two issues:

- the adresses should be hexadecimal (i.e. 1000 -> 0x1000), otherwise these are decimal values, which means wrong adresses 
- the default baud rate is 115200 and not 921600

> How can I monitor logs? I have used Arduino IDE for seeing the logs or sent them by mqtt.

You need a serial monitor tool, the Espressif framework has some built in ([s. here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html)), but you have to setup whole ESP-IDF SDK. An other option is Putty, like described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html)

backupandforkacc, 2023-04-29:
I flashed it again with the correct adresses. I think it did it correctly though as I used esptool before. 
I am using putty to monitor, but only see this (pressed the reset button twice to test it)

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

```

I moved it around without my Iphone in my car for a while to catch some reports, but nothing happened.

beckmx, 2023-04-29:
I think you need a little more experience with esp32, I have flashed several without issues. What you see right there is the state of the esp32 ready to receive a firmware, it looks from my experience a custom board? Which gpio0 kept on low

dchristl, 2023-04-30:
@beckmx is right, this means your flashing doesn't work, but the device is ready for it. A valid output will be something like that:

> headless_haystack: Returned from light sleep
headless_haystack: Current cycle is 2. Reusing key.
headless_haystack: Loading key with index 12 at address XXX
headless_haystack: using key with start XX fXX
headless_haystack: using device address: XX XX XX XX XX XX
headless_haystack: Sending beacon (with key index 12)
headless_haystack: advertising has started.
headless_haystack: stop adv successfully
headless_haystack: Going to sleep

I've tried it out in my virtual machine, with the latest binaries and a newly generated key.  It works at first try with the newest ESP Flash Tool and this settings:

[image]

You see also the output. The whole flashing lasts appr. 30s. After this the output of Putty is something like this:

[image]

So this seems not a problem with the firmware, Windows or the ESP Download Tool. You can try two things. Update the tool and/or erase the whole flash before installing the firmware.

backupandforkacc, 2023-05-15:
I bought a new ESP32, flashed it, and this morning it worked. Now I get this error:

```
Exception happened during processing of request from ('172.17.0.1', 59418)
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 293, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/BaseHTTPServer.py", line 340, in handle
    self.handle_one_request()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/BaseHTTPServer.py", line 328, in handle_one_request
    method()
  File "./FindMy_proxy.py", line 63, in do_POST
    result = json.loads(res.read())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded
----------------------------------------

```

backupandforkacc, 2023-05-15:
How do I login to the virtual mac machine?

This morning, with "docker start -ai ...", I was automatically logged in as:
`user@users-iMac-Pro`

Then I stopped the docker container, restarted it and tried to login:
```
$ docker exec -it simon-headless bash
[arch@86532bce8abe OSX-KVM]$
```

So it seems I was only logged in the container, but not the OS system.
I tried with SSH, but failed.

On the docker host machine, I tried:
```
simon@simon-sff:~$ ssh user@192.168.103.34:50922
ssh: Could not resolve hostname 192.168.103.34:50922: Name or service not known
```
And in the container with:
```
$ docker exec -it simon-headless bash
[arch@86532bce8abe OSX-KVM]$ ssh user@localhost
Warning: Permanently added 'localhost' (ED25519) to the list of known hosts.
user@localhost's password: 
Permission denied, please try again.
```
I tried "alpine" as password.

dchristl, 2024-01-17:
This issue is not valid anymore with the latest release.

closed 2024-01-17 by dchristl