dchristl/macless-haystack #8: FindMy_proxy.py API question

mrmay-dev, 2023-04-30:
I'm working on an OwnTracks integration.  I can get the standalone `report_request.py` to work. I run a Bash script every 20 minutes to trigger a request. 

I'd really like to use the `FindMy_proxy.py` If you could help with the API a little I'd be grateful. As far I can tell it expects a POST request with a JSON body. My guess is, the script expects a list of keys. Does it want the private keys or the prefixes?

```json
{
 "days": 1,
 "ids": [
  "key"
  "key"
 ]
}
```

dchristl, 2023-05-01:
Hello @mrmay-dev ,

you are correct with the API and the body. As key you need the advertisement key, hashed with sha256 in Base64. If you have generated the keys with my generate_keys.py - file, you will find a key-file with all the informations you need. The correct one is the one, that is tagged with "Hashed adv key".

Systm21, 2023-08-27:
Hey richard, really nice project, hope you stick with it.  Will try it in the next days, hopefully it is the "bridge"  to home assistant, that im searching for.

Systm21, 2023-08-31:
@mrmay-dev why are issues deactivated in your owntracks repo? Would love to use your owntracks plugin,  but there seems to be some mistakes in the scripts, that i cant't resolve.  Maybe you can help me a bit.

23tom2323, 2023-09-09:
> @mrmay-dev why are issues deactivated in your owntracks repo? Would love to use your owntracks plugin, but there seems to be some mistakes in the scripts, that i cant't resolve. Maybe you can help me a bit.

I think that's a shame too.
What errors do you get? This comes to me:

./downtags.sh
Traceback (letzter Aufruf zuletzt):
Datei „/request_reports.py“, Zeile 18, in
OUTPUT_FOLDER = Configuration[“owntag_options“][“output_folder "]
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'output_folder'
– END SCRIPT –

python3 request_reports.py --time 0:60 --owntags
Traceback (letzter Aufruf zuletzt):
Datei „/request_reports.py“, Zeile 18, in
OUTPUT_FOLDER = Konfiguration["owntag_options"]["output_folder"]
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^ ^^^^^^^^^
KeyError: 'output_folder'

Meine request_reports.py Zeile 18:
OUTPUT_FOLDER = Configuration["owntag_options"]["output_folder"]

23tom2323 @ sysponet .de

Systm21, 2023-09-09:
> I think that's a shame too. What errors do you get? This comes to me:
>
> ./downtags.sh Traceback (letzter Aufruf zuletzt): Datei „/request_reports.py“, Zeile 18, in OUTPUT_FOLDER = [...]

I have fixed many problems, yours belongs to a var that didn't exists. But now, i get stuck.

mrmay-dev, 2023-09-10:
They are deactivated because I'm a gitHub noob who didn't know they were deactivated. I'm working on figuring that out now. Hope to chat more about this there!

mrmay-dev, 2023-09-10:
I'm looking into the `OUTPUT_FOLDER = Konfiguration["owntag_options"]["output_folder"]` error.

Follow progress, or add more info at the new [issue topic](https://github.com/mrmay-dev/owntags/issues/3#issue-1889218963).