dchristl/macless-haystack #10: generate_keys.py creates non_verifiable keys

aircable, 2023-05-13:
I'm using the generate_keys.py to make advertisement keys for the ESP32.
Inside the ESP32 I verify the public key with the code from https://github.com/positive-security/send-my
I read the keys in from the FILE.keys and call: is_valid_pubkey( key_pointer ).
About 50% of the keys do not verify. And furthermore the sequence:
```
    set_addr_from_key( rnd_addr, key_pointer );
    memcpy( settingsV.adv_data, raw_adv_data, sizeof( raw_adv_data ));
    set_payload_from_key( settingsV.adv_data, key_pointer );
    set_advertising( rnd_addr );
```
often results in a fail setting the random address due to address errors (like C0000000000)

It would be nice to include a public key verification function during generation similar
to the test for '/' in the hashed key.

dchristl, 2023-05-13:
Hello @aircable ,

thank you for your hints, but I don't think I understand it completely. Maybe you can give me more insights. What does 'non verifiable keys' mean in the result, and what is the verification needed for? If the data is not transmitted, will the device be recognized as invalid by Apple? I have patched several devices before and never had any issues with them not being detected. Usually, I only use one key, and if half of it doesn't work, I would have noticed that normally.

I will be happy to add such a test, of course, once I have fully understood the problem.

aircable, 2023-05-13:
Here is an example of a key that causes an error:
`I (63421590) AIRTAG: using device address: c0 00 00 00 00 00
E (63421590) BT_BTC: Invalid static random address, the high bit should be 0b11, bits of the random part shall not be all 1 or 0`
This is the algorithm I use to verify the key:
`
bool 
is_valid_pubkey(uint8_t *pub_key_compressed) {
   uint8_t with_sign_byte[29];
   uint8_t pub_key_uncompressed[128];
   const struct uECC_Curve_t * curve = uECC_secp224r1();
   with_sign_byte[0] = 0x03; // can be 0x02 or 0x03
   memcpy( &with_sign_byte[1], pub_key_compressed, 28 );
   uECC_decompress( with_sign_byte, pub_key_uncompressed, curve );
   if(!uECC_valid_public_key(pub_key_uncompressed, curve)) {   
            return 0;
   }
   ESP_LOGW(TAG, "Key is valid");
   return 1;
}
`
This means that the key isn't a valid point on the curve.
Can you check please?
I use the Android app AirGuard: https://github.com/seemoo-lab/AirGuard
And the airtag with the key only shows up as an Apple Beacon, not an Apple Airtag.
Try it.
What can I do to verify that it is working anyway?

dchristl, 2023-05-14:
Thank you for your answer. I think I understand the problem a little bit better, but not completely. The part with the random address is definitively an issue to solve. 
I used AirGuard before and some of my devices are shown as Beacons, too. But, the devices are functioning normally and transmitting the key normally, and they can be located.

I will debug the code a little bit and will do mny best to solve this.

dchristl, 2023-05-14:
I've added your validation code in a different branch (s. [here](https://github.com/dchristl/headless-haystack/blob/non_verifiable_keys/firmware/ESP32/src/openhaystack_main.c)) and called it before setting the address. I've generated over 1000 keys and none of them was invalid. The keys were generated with my [python file](https://github.com/dchristl/headless-haystack/blob/non_verifiable_keys/webserver/generate_keys.py). Have you changed something in the generation or firmware code (your code fragment from the 1st code differs from mine)? If so, please post your whole code, because I don't see any problem with my sources.

aircable, 2023-05-14:
Your code is fine, no difference. No idea why it's given me unverifiable keys.
I created some my way. Please check the keyfile here with your code. See something wrong?
https://drive.google.com/file/d/1j-yErzCeytgHTWW--SGxJys7W7G9hO0s/view?usp=share_link

dchristl, 2023-05-15:
All keys seems fine, see attached file. 

[keys.log](https://github.com/dchristl/headless-haystack/files/11475714/keys.log)

Maybe you can compare with your output this way.

aircable, 2023-05-17:
thanks for testing them, I'll show you my output soon. I'm traveling, so it might take a moment. Cheers!

dchristl, 2023-05-26:
Hello @mrx23dot ,

I don't understand your point, because everything works as expected. By the way the advertisement key is a hashsed random secp224r1 public key. So it will be really hard to generate a key with "Apple header". What should this be, btw.? I never heard of this and digged very deep into this technology.

dchristl, 2023-06-02:
I get 20 reports with your output, but none of them is valid. They all have the same coordinates of lat: 53.6656128, lon: 426.8757582 . The Longitute have to be between -90 and 90.

dchristl, 2023-06-02:
Can you please provide a complete calculation process? I'm getting completely different numbers (i.e. 1⁷ = 1?). When are you in decimal, when in hexadecimal, ...? Then I might be able to provide a workaround.

dchristl, 2023-06-03:
Thanks for pointing this out and your explanation. I've released a new version with a fix for that.
I will now close the issue since the root problem has been resolved and your issue is unrelated to it. If any further problems arise, please open a new issue.

closed 2023-06-03 by dchristl