dchristl/macless-haystack #15: Any chance for ST17H66 support?

johnbaker26222, 2023-05-30:
These are so cheap and easily available trackers, would be real nice to also support as well.
Thanks!

dchristl, 2023-06-02:
Hello @johnbaker26222,
because of the support provided by [Biemster](https://github.com/biemster/FindMy/tree/main/Lenze_ST17H66), this project can also handle it, because it is compatible. 

Btw. what do you mean by cheap and easily available? The ST17H66 is 3 times the price of an ESP32 C3 and there is only one seller. 

Regards,
Danny

Cyl0nius, 2023-07-30:
The ST17H66 is 3 times the price of an ESP32 C3 and there is only one seller (sic!)

I don't know where you buy your SOC, but I only pay 50 Euro-cents for an ST17H66 and I can't get an ESP32-C3 under 1 USD.
![esp3
[image]
2c3](https://github.com/dchristl/headless-haystack/assets/119928546/c0cc93e3-bfeb-4b8d-a5c8-5d5513f721bb)

johnbaker26222, 2023-07-30:
> [esp3 st17h66 2c3](https://user-images.githubusercontent.com/119928546/257043410-c0cc93e3-bfeb-4b8d-a5c8-5d5513f721bb.jpg)

I get mine here for less than 1USD after you convert to the Thai Baut. https://s.lazada.co.th/s.9M5MC

Then after programming I dremel off everything except the ST17H66 and expoxy it on top of one of these (I need smallest form possible)
https://s.lazada.co.th/s.9M5Fn
And solder a 30mm wire on the antenna pad. It seems to work better than the on board PCB antenna

johnbaker26222, 2023-07-30:
Sorry should have said dremel off everything except the ST17H66 and the crystal

Cyl0nius, 2023-07-31:
@johnbaker26222 

Can you please post a picture of the reduced (dremeled) board?

johnbaker26222, 2023-07-31:
[image]

dchristl, 2023-07-31:
> I don't know where you buy your SOC, but I only pay 50 Euro-cents for an ST17H66 and I can't get an ESP32-C3 under 1 USD.

 It seems you are getting different results than me. Furthermore, you are comparing a SoC with a development board. My prices are (on aliexpress Germany) [St17h66 for 13,29€](https://de.aliexpress.com/item/1005004766401470.html?spm=a2g0o.productlist.main.1.350b1e25NrY7ox&algo_pvid=2cf9e416-3584-442b-87c1-98b52eb413a0&algo_exp_id=2cf9e416-3584-442b-87c1-98b52eb413a0-0&pdp_npi=3%40dis%21EUR%2113.29%2113.29%21%21%2114.30%21%21%40211b88ee16908067101037568e5194%2112000030392369327%21sea%21DE%213120182161&curPageLogUid=pQVMNRvRAZBi) and [Esp32 for 4,41€](https://de.aliexpress.com/item/1005005626327947.html?spm=a2g0o.productlist.main.11.505332efOuaKXs&algo_pvid=679cda94-6638-4b7b-b649-00f03ceada07&algo_exp_id=679cda94-6638-4b7b-b649-00f03ceada07-5&pdp_npi=3%40dis%21EUR%216.58%214.41%21%21%217.08%21%21%40211b88ee16908071977467767e5191%2112000033794793676%21sea%21DE%213120182161&curPageLogUid=9LlOl5Zrm5TL) Maybe you see other prices at your location. 

> And have you ever thought about the power consumption?

I'm aware of that. 

In general, I find the idea of a portable battery-operated chip very appealing, but I don't want to create an easily usable stalking device. 

I would be interested in delving into the topic as well, but documentation, examples, frameworks, etc. are hard to come by for the chip. If someone could provide me with instructions for flashing (pinout, software, flash tool), I'd be happy to explore it further.

@johnbaker26222: Do you use Biemster's software with headless haystack?

Systm21, 2023-08-03:
I don't know why @Cyl0nius  takes it the wrong way, because he himself suggested the chip to Biemster in December and helped examine it.

Here are some links that should shed some light: 
https://github.com/biemster/FindMy/tree/main/Lenze_ST17H66
https://github.com/biemster/FindMy/issues/14

Systm21, 2023-08-22:
@dchristl  any chance to support these little things? They're really Cheap (Auch in Deutschland, z.B. als Schlüsselfinder für 2-3€). so that would be awesome.

Cyl0nius, 2023-08-23:
@ Systm21 ... is already available at: https://github.com/biemster/FindMy

Cyl0nius, 2023-08-23:
@Systm21 .. what kind of support you are asking for?
In almost all of the key finders you mentioned, an ST17H66 is installed.

dchristl, 2023-08-23:
Hello @Systm21 ,

like I mentioned before, these chips (ST17H66) are already supported, in this case you can find the firmware and the instructions [here](https://github.com/biemster/FindMy/tree/main/Lenze_ST17H66) 
In general, this project also supports all trackers that are supported by OpenHaystack. For flashing, you usually only need the advertisement key, which you can obtain through generate_keys (s. [instructions](https://github.com/dchristl/headless-haystack#instructions)). Then you will have a file with ending keys with all you need, i.e:
[image]

 The correct one is the **Advertisement key**, that is encoded in base64. For this example and the ST17H66 you can directly flash with
`./flash_st17h66.py gYZSzPhWRmR3Ucfbo021mcChPsTQ7JpUh/W5Hg==`

The only thing that doesn't work is the "rotating key" feature. This is used whenever more than 1 key is generated. The key then rotates every 30 minutes on the chip. This needs to be implemented/added by me.

I have already ordered a few of the chips and a programmer and have done some reading. The entire programming and flashing process appears to be quite fragile and experimental (and Windows-only). Documentation also seems to be available only in Chinese. Therefore, I currently have little hope of adding this feature. I'll know more once my order arrives.

If you need a similar portable tracker with the "rotating key" feature, you can also try the NRF51. I added this chip and functionality the day before yesterday. The instructions are located [here](https://github.com/dchristl/headless-haystack/blob/main/firmware/nrf5x/README.md)

Regards,
Danny

Systm21, 2023-08-23:
@Cyl0nius 
I don't know if you being ironic, but of course not every keychain of this type automatically houses an ST17H66.

@ddchristel
Of course, biemster delivered a fully functional solution, but rolling key would be a breaking change.

dchristl, 2023-08-24:
Why is this a breaking change, besides the fact that stalking is possible with that. What other use cases do you have?

Systm21, 2023-08-24:
> Why is this a breaking change, besides the fact that stalking is possible with that. What other use cases do you have?

https://github.com/dchristl/headless-haystack/issues/21

johnbaker26222, 2023-08-25:
I don't think the key rotation is necessary. I've got plenty of these devices and have taken them on many trips an never received a warning from my iphone, ipad, or mac that a device was nearby. I suspect it has to do with the lower transmission rate than the airtags that come from apple. There are many great uses for these other than stalking. I just shipped a valuable package internationally. I used a cut down one on top of a CR1220. It was so small I jammed it in between the corrugated cardboard. Even got location updates in the air. So I suspect it was loaded in the cargo compartment on a passenger flight that had Wifi? Because of their low cost I don't mind at all never getting them back. Get to see where the dump is in various countries.
Also with young kids, I have these inside their shoe soles. Also have a script that notifies me when a location is moved X number distance via telegram. These are probably the smallest, longest lasting trackers in the world. Many many uses for them.

Systm21, 2023-08-25:
@johnbaker26222 
Sounds really cool, how do you make the connection between Telegram and the Findmy network? 
my son recently started school and wears one around his neck and the other is attached to his school bag, would like to automare it as you described.  Thought about doing it with Homeassistent.

Systm21, 2023-08-25:
@dchristl 
sorry the linked issue didn't described your question really well. I know there is a indoor tracking system in my sons school. Normally this was used in covid times to track if there are many peoples in the hallways. Its still active and rolling keys are blocking the exact tracking. Thats the reason, why "normal" airtags are also rolling the keys (5 or 6?!) . Its not only the school, which is tracking. So when my son is wearing it, i would feel better, when he is not getting tracked so easy.

johnbaker26222, 2023-08-25:
Its just a simple python script I have running on the same server as headless haystack. Put it in the same folder as request_reports.py is in. Put your generated keys of the devices you want the alerts for. If you used the openhaystack app to create the devices and don't have the keypairs, then copy in the base64 priv key in this slightly modified script below to generate what you need. Just google around on how to get the telegram bot token and chat id. 

```#!/usr/bin/env python3
import time
import subprocess
import json
from geopy.distance import geodesic
from telegram import Bot

# Telegram bot token - Replace 'YOUR_TELEGRAM_BOT_TOKEN' with your actual bot token.
telegram_bot_token = '4018577110:AAFkh0ktYlB4BcsMHGUE5PzZD-J6aewDeIk'

# Telegram chat ID - Replace 'YOUR_TELEGRAM_CHAT_ID' with your actual Telegram chat ID.
telegram_chat_id = '5144424153'

# Dictionary to store the last known location of each key.
last_known_locations = {}

# Function to get the distance between two coordinates in miles.
def get_distance(coord1, coord2):
    return geodesic(coord1, coord2).miles

# Function to send a Telegram alert.
def send_telegram_alert(key, distance):
    bot = Bot(token=telegram_bot_token)
    message = f"Item '{key}' has moved {distance:.2f} miles."
    bot.send_message(chat_id=telegram_chat_id, text=message)

# Main loop to run the geofencing check every 5 minutes.
while True:
    try:
        print("Running the geofencing check...")

        # Execute the request_reports.py script and capture the output. Replace with your icloud key. Use $ security find-generic-password -ws 'iCloud' 
        process = subprocess.Popen(['python', 'request_reports.py', '-H', '1', '-k', '9/Q4Cwq1QnNUeimNBMI5B3boTuO+7l2NLLM7HvmWViQ='], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
        stdout, stderr = process.communicate()

        # Decode the output and split it into individual lines.
        output_lines = stdout.decode().splitlines()

        # Parse the output to extract location data for each key.
        location_data = [json.loads(line.replace("'", '"')) for line in output_lines if line.startswith('{')]
        for data in location_data:
            key = data['key']
            latitude = float(data['lat'])
            longitude = float(data['lon'])
            timestamp = int(data['timestamp'])

            # Calculate the distance between the current location and the last known location.
            current_location = (latitude, longitude)
            last_location = last_known_locations.get(key)
            if last_location:
                distance_miles = get_distance(current_location, last_location)
                print(f"Checking '{key}' - Distance: {distance_miles:.2f} miles")
                if distance_miles > 1.0:     #Adjust this for the distance trigger sorry I think in miles :(
                    send_telegram_alert(key, distance_miles)

            # Update the last known location for the key.
            last_known_locations[key] = current_location

        print("Geofencing check completed.")

    except Exception as e:
        print("Error:", e)

    # Wait for 5 minutes before checking again.
    print("Waiting for the next geofencing check...")
    time.sleep(5 * 60)```

Modified generate_keys.py 
```#!/usr/bin/env python2
import sys,base64,hashlib,random
from cryptography.hazmat.primitives.asymmetric import ec
from cryptography.hazmat.backends import default_backend
import argparse
import codecs,struct

def int_to_bytes(n, length, endianess='big'):
    h = '%x' % n
    s = ('0'*(len(h) % 2) + h).zfill(length*2).decode('hex')
    return s if endianess == 'big' else s[::-1]

def sha256(data):
    digest = hashlib.new("sha256")
    digest.update(data)
    return digest.digest()

def bytes_to_int(b):
    return int(codecs.encode(b, 'hex'), 16)

parser = argparse.ArgumentParser()
parser.add_argument('-n','--nkeys', help='number of keys to generate', type=int, default=1)
parser.add_argument('-p','--prefix', help='prefix of the keyfiles')
parser.add_argument('-y','--yaml', help='yaml file where to write the list of generated keys')
parser.add_argument('-v','--verbose', help='print keys as they are generated', action="store_true")
args = parser.parse_args()

if args.yaml:
    yaml=open(args.yaml + '.yaml','w')
    yaml.write('  keys:\n')

for i in range(args.nkeys):
    #priv = random.getrandbits(224)
    priv = bytes_to_int(base64.b64decode('c2g0uUGTyxE43BBjZ7gAL2oq5cBZkmet+j4R1w=='))  #Copy your priv key in base 64 here from openhaystack to generate the key pair needed. 
    adv = ec.derive_private_key(priv, ec.SECP224R1(), default_backend()).public_key().public_numbers().x

    priv_bytes = int_to_bytes(priv, 28)
    adv_bytes = int_to_bytes(adv, 28)

    priv_b64 = base64.b64encode(priv_bytes).decode("ascii")
    adv_b64 = base64.b64encode(adv_bytes).decode("ascii")
    s256_b64 = base64.b64encode(sha256(adv_bytes)).decode("ascii")

    if args.verbose:
        print('%d)' % (i+1))
        print('Private key: %s' % priv_b64)
        print('Advertisement key: %s' % adv_b64)
        print('Hashed adv key: %s' % s256_b64)

    if '/' in s256_b64[:7]:
        print('no key file written, there was a / in the b64 of the hashed pubkey :(')
    else:
        if args.prefix:
            fname = '%s_%s.keys' % (args.prefix, s256_b64[:7])
        else:
            fname = '%s.keys' % s256_b64[:7]

        with open(fname, 'w') as f:
            f.write('Private key: %s\n' % priv_b64)
            f.write('Advertisement key: %s\n' % adv_b64)
            f.write('Hashed adv key: %s\n' % s256_b64)

        if args.yaml:
            yaml.write('    - "%s"\n' % adv_b64)```

johnbaker26222, 2023-08-25:
Modified generate_keys.py 
```#!/usr/bin/env python2
import sys,base64,hashlib,random
from cryptography.hazmat.primitives.asymmetric import ec
from cryptography.hazmat.backends import default_backend
import argparse
import codecs,struct

def int_to_bytes(n, length, endianess='big'):
    h = '%x' % n
    s = ('0'*(len(h) % 2) + h).zfill(length*2).decode('hex')
    return s if endianess == 'big' else s[::-1]

def sha256(data):
    digest = hashlib.new("sha256")
    digest.update(data)
    return digest.digest()

def bytes_to_int(b):
    return int(codecs.encode(b, 'hex'), 16)

parser = argparse.ArgumentParser()
parser.add_argument('-n','--nkeys', help='number of keys to generate', type=int, default=1)
parser.add_argument('-p','--prefix', help='prefix of the keyfiles')
parser.add_argument('-y','--yaml', help='yaml file where to write the list of generated keys')
parser.add_argument('-v','--verbose', help='print keys as they are generated', action="store_true")
args = parser.parse_args()

if args.yaml:
    yaml=open(args.yaml + '.yaml','w')
    yaml.write('  keys:\n')

for i in range(args.nkeys):
    #priv = random.getrandbits(224)
    priv = bytes_to_int(base64.b64decode('c2g0uUGTyxE43BBjZ7gAL2oq5cBZkmet+j4R1w=='))  #Copy your priv key in base 64 here from openhaystack to generate the key pair needed. 
    adv = ec.derive_private_key(priv, ec.SECP224R1(), default_backend()).public_key().public_numbers().x

    priv_bytes = int_to_bytes(priv, 28)
    adv_bytes = int_to_bytes(adv, 28)

    priv_b64 = base64.b64encode(priv_bytes).decode("ascii")
    adv_b64 = base64.b64encode(adv_bytes).decode("ascii")
    s256_b64 = base64.b64encode(sha256(adv_bytes)).decode("ascii")

    if args.verbose:
        print('%d)' % (i+1))
        print('Private key: %s' % priv_b64)
        print('Advertisement key: %s' % adv_b64)
        print('Hashed adv key: %s' % s256_b64)

    if '/' in s256_b64[:7]:
        print('no key file written, there was a / in the b64 of the hashed pubkey :(')
    else:
        if args.prefix:
            fname = '%s_%s.keys' % (args.prefix, s256_b64[:7])
        else:
            fname = '%s.keys' % s256_b64[:7]

        with open(fname, 'w') as f:
            f.write('Private key: %s\n' % priv_b64)
            f.write('Advertisement key: %s\n' % adv_b64)
            f.write('Hashed adv key: %s\n' % s256_b64)

        if args.yaml:
            yaml.write('    - "%s"\n' % adv_b64)```

dchristl, 2023-08-26:
> I don't think the key rotation is necessary. I've got plenty of these devices and have taken them on many trips and have never received a warning from my iphone, ipad, or mac that a device was nearby. I suspect it has to do with the lower transmission rate than the airtags that come from apple. There are many great uses for these other than stalking. I just shipped a valuable package internationally. I used a cut down one on top of a CR1220. It was so small I jammed it in between the corrugated cardboard. Even got location updates in the air. So I suspect it was loaded in the cargo compartment on a passenger flight that had Wifi? Because of their low cost I don't mind at all never getting them back. Get to see where the dump is in various countries.
> Also with young kids, I have these inside their shoe soles. Also have a script that notifies me when a location is moved X number distance via telegram. These are probably the smallest, longest lasting trackers in the world. Many many uses for them.

@johnbaker26222 
Thank you for your answer and I'm aware of this use cases. This already works as well, even now. My question about the use cases was specifically directed towards the feature involving the changing keys. ;)

> sorry the linked issue didn't described your question really well. I know there is a indoor tracking system in my sons school. Normally this was used in covid times to track if there are many [...]

@Systm21 
This is very special and kind of weird school if they track there students. ;) Are you sure that normal airtags rolling the keys, too? I never read anything about it. As mentioned before, I cannot currently guarantee that I can implement this at all. I had already provided the reasons. Currently, you can only solve this by using an Nrf51 (the instructions also contain links to the trackers) and generating 50 keys. The Nrf51 is only slightly more expensive than the Sth1766.

Systm21, 2023-08-26:
@dchristl they are not tracking their students, only the people in some specific areas. Hallways, Theatre etc. 
Dont know if it is used anymore, during covid, it was not that bad.
Anyway, also if it is not bundled with a student, i dont want to have this "can" situation.

Cyl0nius, 2023-08-27:
@johnbaker26222
My trackers (STt17h66) run internationally. 
A script runs in a VM that retrieves the position data at definable intervals and displays it on a website. 

I have repeatedly observed that a position report is made during intercontinental flights.
It looks like it's always in the same place. I will keep watching.

[image]

dchristl, 2023-08-27:
@Systm21 
I'm sorry, even though this is somewhat off-topic, I don't quite understand how this is supposed to work. So how are people tracked? The private key of the tracker is unknown, so the data cannot be determined via the FindMy network. Are all Bluetooth devices tracked via MAC addresses? If so, doesn't your son have a phone, wearable, Bluetooth headphones, etc., which are much easier to track? There are no changing MAC addresses.

Systm21, 2023-08-27:
@dchristl good point, didn't thought about, that tracking is made with the mac 🤔. 
I did a little research, you're right, but some smartphones can actually refresh the mac regularly.

Cyl0nius, 2023-09-17:
@johnbaker26222

I will try now a helical antenna around the (reduced) pcb with hearing-aid batteries on both sides ... should end in a nice small cylinder form factor.

Do you have any experience with antennas to share?

[image]

[image]

johnbaker26222, 2023-09-17:
Cool! No all I use is a short 25-30 mm wire antenna soldered directly to the unused pad. It seems to work better than the ones I left the PCB antenna attached.
Also I always program them before I cut them. Then you can cut the legs on the far side of the chip off as well making it a bit smaller. I'm tempted to make some custom boards with programming pads in the opposite side with the crystal right next to the ST for the absolute smallest form.

biemster, 2023-09-21:
> I get mine here for less than 1USD after you convert to the Thai Baut. https://s.lazada.co.th/s.9M5MC Then after programming I dremel off everything except the ST17H66 and expoxy it on top of one of [...]

That looks awesome!

biemster, 2023-09-21:
> Are you sure that normal airtags rolling the keys, too? I never read anything about it. As mentioned before, I cannot currently guarantee that I can implement this at all. I had already provided the [...]

Yes the normal airtags are rolling the keys as well, to prevent MAC address tracking (part of the public key is in the last bytes of the bluetooth mac).

I understand your preference for the nRF series, but qua price the st17h66 is unbeatable. With a bit of aliexpress skills you can get at least 3 tags with this chip for the price of a single nrf, so if you are planning to send out a whole fleet of those (as apparently some of us do, including me), the st17h66 is unavoidable.
Not to mention the nice little case most of them come in, I did not find an nRF in small keychain form factor yet.

Anyway this is just my opinion of course, and the reason I focus in my project on this chip. And it's great to see that you picked some of that up and made it even awesomer! thanks!

dchristl, 2023-09-22:
Hello @biemster,

nice to hear from you.

> This is painting a bit too grim picture! The firmware code can be compiled with GCC, and the flasher is just a python script. All development on this is done on Linux, so that's a first class citizen on this.
The fragility comes mostly from the chip itself, not the support. Documentation is in English and plenty, as are SDK examples.
Adding rolling keys is high on my list of implementation (https://github.com/biemster/st17h66_FindMy/issues/2), and actually quite easy (just did not find the time yet). Exactly for the same reason of MAC address tracking, but since I still have to carry a smartphone often it is of course a bit moot.

I haven't really delved deep into the topic yet and have only scratched the surface. However, the chips have arrived, and I will be delving deeper into it soon. My benchmark is the excellent documentation of the ESP32, and I didn't find much at first glance. But I'm glad to hear that your Makefile runs smoothly with GCC on Linux. That aligns with my setup as well. The issue with the available time is my problem, too ;)

> Yes the normal airtags are rolling the keys as well, to prevent MAC address tracking (part of the public key is in the last bytes of the bluetooth mac).

Are you sure that the key and the Mac are being swapped? I can understand the Mac being swapped, but the key? How does Apple then implement their anti-stalking feature on the iPhone?

> I understand your preference for the nRF series, but qua price the st17h66 is unbeatable. With a bit of aliexpress skills you can get at least 3 tags with this chip for the price of a single nrf, so [...]
Not to mention the nice little case most of them come in, I did not find an nRF in small keychain form factor yet.

I don't have a preference for any particular chip, it was just the first firmware I found on a chip with a smaller form factor, and it seems to be the chip Apple uses in the AirTag as well. The STh1766 is undoubtedly much better, I only discovered it too late. And since you've given me some hope for flashing and programming, it definitely sounds much better than I had thought.

> And it's great to see that you picked some of that up and made it even awesomer! thanks!

I can return the thanks. Your web server is very helpful in achieving my (original) goal of running the entire OpenHaystack project without a physical Mac, yet still having a graphical UI.

Kind regards,
Danny

Systm21, 2023-09-22:
> Its just a simple python script I have running on the same server as headless haystack. Put it in the same folder as request_reports.py is in and also put your generated keys of the devices you want the alerts for all in the same folder. If you used the openhaystack app to create the devices like I do and don't have the key pairs, then copy in the base64 private key in this slightly modified script below to generate what you need. You probably want to rename the keys to what device they belong to so the telegram alert makes more sense.
> 
> Just google around on how to get the telegram bot token and chat id.
> 
> ```
> #!/usr/bin/env python3
> import time
> import subprocess
> import json
> from geopy.distance import geodesic
> from telegram import Bot
> 
> # Telegram bot token - Replace 'YOUR_TELEGRAM_BOT_TOKEN' with your actual bot token.
> telegram_bot_token = 'YOUR_TELEGRAM_BOT_TOKEN'
> 
> # Telegram chat ID - Replace 'YOUR_TELEGRAM_CHAT_ID' with your actual Telegram chat ID. (include - in front of numbers if it exists)
> telegram_chat_id = 'YOUR_TELEGRAM_CHAT_ID'
> 
> # Dictionary to store the last known location of each key.
> last_known_locations = {}
> 
> # Function to get the distance between two coordinates in miles.
> def get_distance(coord1, coord2):
>     return geodesic(coord1, coord2).miles
> 
> # Function to send a Telegram alert.
> def send_telegram_alert(key, distance):
>     bot = Bot(token=telegram_bot_token)
>     message = f"Item '{key}' has moved {distance:.2f} miles."
>     bot.send_message(chat_id=telegram_chat_id, text=message)
> 
> # Main loop to run the geofencing check every 5 minutes.
> while True:
>     try:
>         print("Running the geofencing check...")
> 
>         # Execute the request_reports.py script and capture the output. Replace with your icloud key. Use $ security find-generic-password -ws 'iCloud' 
>         process = subprocess.Popen(['python', 'request_reports.py', '-H', '1', '-k', 'YOUR_ICLOUD_KEY'], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
>         stdout, stderr = process.communicate()
> 
>         # Decode the output and split it into individual lines.
>         output_lines = stdout.decode().splitlines()
> 
>         # Parse the output to extract location data for each key.
>         location_data = [json.loads(line.replace("'", '"')) for line in output_lines if line.startswith('{')]
>         for data in location_data:
>             key = data['key']
>             latitude = float(data['lat'])
>             longitude = float(data['lon'])
>             timestamp = int(data['timestamp'])
> 
>             # Calculate the distance between the current location and the last known location.
>             current_location = (latitude, longitude)
>             last_location = last_known_locations.get(key)
>             if last_location:
>                 distance_miles = get_distance(current_location, last_location)
>                 print(f"Checking '{key}' - Distance: {distance_miles:.2f} miles")
>                 if distance_miles > 1.0:     #Adjust this for the distance trigger sorry I think in miles :(
>                     send_telegram_alert(key, distance_miles)
> 
>             # Update the last known location for the key.
>             last_known_locations[key] = current_location
> 
>         print("Geofencing check completed.")
> 
>     except Exception as e:
>         print("Error:", e)
> 
>     # Wait for 5 minutes before checking again.
>     print("Waiting for the next geofencing check...")
>     time.sleep(5 * 60)'
> ```

I have tryed to make it run, but it seems there is something wrong with the formatting of the json string. It can't extract the key, longitude, latitude and timestamp variables from the json data. Can you review it and help me out?

biemster, 2023-09-22:
@Systm21 a copy of the error message would definitely help!

Systm21, 2023-09-22:
> @Systm21 a copy of the error message would definitely help!

`Error: 'key'`

Systm21, 2023-09-22:
> @Systm21 a copy of the error message would definitely help!

`Error: key`

When i print out what comes out of  'data', it's clearer what the problem is, but i don't know how to solve it. 

`{'search': [{'endDate': 717115914000000, 'startDate': 717043914000000, 'ids': ['ERreQEFweEwffWMez6WCWIJAZv534§3rfdsFdFDdc=']}]}`

biemster, 2023-09-22:
your request_reports.py script seems to contain some debug print statement, where it prints the line that it is sending to the apple servers. can you try and find that line and comment it out?

Systm21, 2023-09-22:
the request_reports.py works as expected, without any issues. It only belongs to the telegram/geofence script from @johnbaker26222.

johnbaker26222, 2023-09-23:
> the request_reports.py works as expected, without any issues. It only belongs to the telegram/geofence script from @johnbaker26222.

You did insert your iCloud key right? security find-generic-password -ws 'iCloud' 

I did have some json parsing issues initially but got them fixed with this script. It works for me ? Really I need to rewrite this. Since it only measures the distance moved every 5 minutes and is triggered by exceeding a distance, what its really doing is measuring a speed which isn't what I want. I think I may update it so it sends an alert when moved a certain distance from a home position. And maybe cancel the alerts if another device I've programed is nearby. This would work good at least for my use case, if my kids are not near me or my wife after say 30 minutes I want an alert.

johnbaker26222, 2023-09-23:
Ohh and no I did not modify the request_reports.py file. Only the generate_keys.py since I create the keys using openhaystack app and need to modify them for the correct format for the request_reports.py

Systm21, 2023-09-23:
> You did insert your iCloud key right? security find-generic-password -ws 'iCloud'

Yes, i insert the key, should i remove it?

> I did have some json parsing issues initially but got them fixed with this script.

Seems to be my problem.

johnbaker26222, 2023-09-23:
Here try this you need to change the telegram token and chat id and your apple ID key, I left examples in (not valid) for reference. The sample.key file is valid though I no longer use it. Try these and let me know if this works.
https://filebin.net/4wz7qzxja7h2p877

dchristl, 2023-09-23:
Quick update on the STH1766. Today, I was able to successfully flash 3 chips with the firmware from biemster as [described](https://github.com/dchristl/headless-haystack/issues/15#issuecomment-1690196976). This works great, and the whole process is easier than I thought. Unfortunately, what didn't work is that when I use the self-compiled result, I can't get tracking to work. 

@biemster 
Is the binary from your repository compiled under Linux with the gcc-arm compiler (my version: arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release))? Do you have any idea where my issue might be? The output generates many warnings. I have attached the output.
[output.txt](https://github.com/dchristl/headless-haystack/files/12706893/output.txt)

Thanks for your help.

biemster, 2023-09-23:
Which code base are you trying to compile? I started a couple side repos with the intention of updating the `FindMy` repo in the end, but just before I would consider everything finished I got distracted by other stuff and never consolidated all the code.
- The https://github.com/biemster/st17h66_FindMy repo has a working FindMy implementation, but still needs to be compiled with Keil.
- The https://github.com/biemster/st17h66_RF repo has working BLE broadcast implementation, proper deep sleep, compiles under GCC (and Keil), but does not have the FindMy key broadcast implemented (should be very easy though: https://github.com/biemster/st17h66_FindMy/commit/daf59e6efa63677289a9a7551be56e47afd7f9ef)

Last thing I was working on (after getting compiling with GCC to work on the _RF repo), was getting the power consumption down (https://github.com/biemster/st17h66_FindMy/issues/7).

I don't think I ever got to combine all efforts (low power, GCC, FindMy bcst), the st17h66_FindMy repo has low power and FindMy, and the _RF repo has low power and GCC support.

dchristl, 2023-09-23:
OK, I've used the https://github.com/biemster/st17h66_FindMy-repo. That's where I found the FindMy implementation and the reference to the GCC compiler. This certainly explains why it's not working. I'll take a look at the repositories tomorrow, but currently, it doesn't look promising if you don't see a way to merge all the features. I'm not nearly as deep into this as you are. Nevertheless, it's a cool project, and thank you for your work. In principle, your binary and Headless-Haystack are compatible with each other.

biemster, 2023-09-23:
> Are you sure that the key and the Mac are being swapped? I can understand the Mac being swapped, but the key? How does Apple then implement their anti-stalking feature on the iPhone?

The [OpenHaystack paper](https://arxiv.org/pdf/2103.02282.pdf) mentions the following at the end of section 6.2:
```
The same key is emitted during a window of 15 minutes, after which the next key
pi+1 is used. 
```
The whole of section 6.2 explains that in order to put a 28 byte payload (public key) in a BLE advertisement packet they had to use some of the MAC address bytes, see also here:
https://github.com/biemster/st17h66_FindMy/blob/68f7b2be9ca8ec68f289e885bdb3673ee2179153/FindMy/source/FindMy.c#L115
so whenever a new key is rolled, the MAC address changes too.

In https://github.com/biemster/st17h66_FindMy/issues/2 there is a bit more discussion on key cycling and how the anti-stalking feature works on an iPhone, and although we are not sure yet it seems that the iPhone looks in this same 15 minute window, and also uses the RSSI of the received FindMy packets.

Systm21, 2023-09-24:
> Here try this you need to change the telegram token and chat id and your apple ID key, I left examples in (not valid) for reference. The sample.key file is valid though I no longer use it. Try these and let me know if this works. https://filebin.net/4wz7qzxja7h2p877

Nice, works now. Had to fiddle around, because its python2 and python3 mixed up, but it is running now. Do you want multiple messages to be sent each time the loop repeats? I always get 4 messages at once, even if the time period is only one hour.

dchristl, 2023-09-24:
@biemster 
Thank you for your informations. I must have overlooked that part in the paper. It's fascinating how Apple implements anti-stalking. I've been assuming all along that changing the key would be enough to bypass anti-stalking, but apparently, it's only the key rotation interval (and possibly the number of keys).

biemster, 2023-09-24:
> > https://github.com/biemster/st17h66_FindMy/blob/68f7b2be9ca8ec68f289e885bdb3673ee2179153/FindMy/source/FindMy.c#L115
> 
> Does this mean the rolling key already works in your firmware, but the flash script takes only one key?

No unfortunately not.
But that block of code does show how easy it would be to implement: expand `static uint8 public_key[]` to hold `n` keys, populate them with the flasher, and call `set_addr_from_key` and `set_payload_from_key` with an index every now and then in `SimpleBLEPeripheral_ProcessEvent`

Cyl0nius, 2023-09-25:
> @biemster Thank you for your informations. I must have overlooked that part in the paper. It's fascinating how Apple implements anti-stalking. I've been assuming all along that changing the key [...]

Changing/rotating public keys is not necessary at all.
Altering the status byte to 0x00, which tricks the victim’s device into believing that the transmissions are from an iPhone and does not trigger an alert [38].

[38] Travis Mayberry, Ellis Fenske, Dane Brown, Jeremy Martin, Christine Fossaceca,
Erik C Rye, Sam Teplov, and Lucas Foppe. 2021. Who Tracks the Trackers? Circumventing
Apple’s Anti-Tracking Alerts in the Find My Network. In Proceedings
of the 20thWorkshop onWorkshop on Privacy in the Electronic Society. ACM, Korea,
181–186. https://doi.org/10.1145/3463676.3485616

I have running a lot of ST17H66, not a single one has ever triggered an alarm.

biemster, 2023-09-25:
@Cyl0nius Interesting! I did not know this. To be fair, I'm mostly worried about others tracking **me** when I carry those tags around and to prevent that I need to cycle the MAC address, but this is very useful info when using the tags as anti-theft devices so they are not easily removed.
I'll move this to the top of things to implement, thanks!

biemster, 2023-10-05:
> This is an awesome project. May I ask if these Tags listed [here](https://github.com/dchristl/headless-haystack/issues/15#issuecomment-1657092629) (i would buy them [here](https://a.aliexpress.com/_mLv9Xqe), think they are the same, right?) work "out of the box" with this project or will I need to solder around and flash them somehow. I am not a talented solderer/coder...

It's a bit of a gamble if you get a tag with an st17h66, you'll have to try and see.. I sometimes get different chips from the same seller. And they for sure will not work out of the box, you'll need to solder around (a bit targeted actually), and flash a custom firmware. With the firmware we can help you, and soldering is a useful skill to have under your belt!

> * Installing Openhaystack in a Docker Container on a Synology NAS

I doubt a NAS is able to run a full os like macOS, you probably need an average desktop/laptop for this

> * using both these super cheap airtag-clone devices linked above and original AirTags without the need to do any soldering

Yeah soldering required, or at least some trickery with probes / test pins.

> * Optimal, yet optional: track these Tags wtih HomeAssistant to automate certain actions

There is nothing stopping you from pushing the reports on an MQTT queue or such, which then should be available in HomeAssistant

> * Optional: if this will not work with HA: automate Telegram bot actions for reaching/leaving predefined areas (at a certain time) e.g. "Your Cat is not at home" after 9pm

I think there is a telegram script a couple posts up in this issue. You mentioned you haven't done any deep reading yet, this is definitely a requirement to get this up and running the way you want.

> * I guess [this](https://www.mokoblue.com/mokoblue-esp32-gateway/) would work as an "at home" receiver, right?

Any bluetooth capable device can receive the BLE advertisements and process them "at home", so this ESP32 thing as well. Also RPi's or any other SBC or even just your home workstation can do this.

> * I do not care too much about the stealth feature (as long as these things dont start making any sound) but would like to test it . Can I change the status byte remotely on [these ](https://a.aliexpress.com/_mLv9Xqe) as described [above](https://github.com/dchristl/headless-haystack/issues/15#issuecomment-1732738154) or would that involve soldering/additional hardware?

This will require an update to the custom firmware you'll need to flash, I did not get to implementing that yet.

> In short my use case is: "Using (Off-Brand) AirTags on Android devices, Bonus: HomeAssistant Integration"

I'm not sure how your Android is coming into this story, but HomeAssistant integration is definitely possible but will require a bit of effort.

dchristl, 2023-10-06:
@biemster 

Just to summarize, hoping I understood it correctly: The most up-to-date functioning firmware is this branch, which was compiled with Keil. All other GCC attempts are not suitable since broadcasting is missing, making it impossible to track. The only realistic chance (without significant modifications) to implement the feature with changing keys is using the (commercial) Keil compiler?

@Cyl0nius 

Thank you for the link/video, this clears a lot how the anti-tracking feature works and how often the key is switched.

@supaeasy 

My response would have been similar to biemster's. In general, anything is possible; your NAS needs to be a real PC. There may be NAS devices on the market that run, for example, Windows, or you can build your own where it might work. Unfortunately, you won't be able to avoid soldering. I conducted my tests with [this](https://de.aliexpress.com/item/1005005834064700.html) and [this device](https://de.aliexpress.com/item/1005004495296995.html). I flashed it with [that](https://de.aliexpress.com/item/1005005925535610.html), if this helps. For an Home-Assistant integration you can have a look at [this project](https://github.com/mrmay-dev/owntags)

By Android, do you mean the APK from this project?

biemster, 2023-10-06:
Hi @dchristl that is almost correct. Currently the only fully functioning FindMy firmware is https://github.com/biemster/st17h66_FindMy, which has to be compiled with Keil indeed. All future development will happen in https://github.com/biemster/st17h66_RF, which is only very marginally behind the aforementioned one and compiles with GCC. The moment I find some time again for this project the `st17h66_RF` repo will be the main one, which really should not take more than a couple hours.

johnbaker26222, 2023-10-16:
Here is a better version of a geofence alert via telegram. This allows a home position to be specified. If a tracker goes outside the fence (1 mile radius in this example) it will send an alert once to prevent nuisance alerts. This should automatically rearm for another alert when the tracker is back inside the fence. 

```#!/usr/bin/env python3
import time
import subprocess
import json
from geopy.distance import geodesic
from telegram import Bot

# Telegram bot token - Replace 'YOUR_TELEGRAM_BOT_TOKEN' with your actual bot token.
telegram_bot_token = 'YOUR_BOT_TOKEN'

# Telegram chat ID - Replace 'YOUR_TELEGRAM_CHAT_ID' with your actual Telegram chat ID.
telegram_chat_id = 'YOUR_CHAT_ID'

# Define your home coordinates here (latitude and longitude).
home_coordinates = (your_home_latitude, your_home_longitude)

# Dictionary to store the last known location of each key and its alert status.
last_known_locations = {}
item_alert_status = {}

# Function to get the distance between two coordinates in miles.
def get_distance(coord1, coord2):
    return geodesic(coord1, coord2).miles

# Function to send a Telegram alert.
def send_telegram_alert(key, distance):
    bot = Bot(token=telegram_bot_token)
    message = f"Item '{key}' has moved {distance:.2f} miles from home."
    bot.send_message(chat_id=telegram_chat_id, text=message)

# Main loop to run the geofencing check every 5 minutes.
while True:
    try:
        print("Running the geofencing check...")

        # Execute the request_reports.py script and capture the output.
        process = subprocess.Popen(['python', 'request_reports.py', '-H', '1', '-k', '7/Q4Cwq1QnNUpimNPMI5B3aoTuO+7l2NLLM7HvmQViQ='], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
        stdout, stderr = process.communicate()

        # Decode the output and split it into individual lines.
        output_lines = stdout.decode().splitlines()

        # Parse the output to extract location data for each key.
        location_data = [json.loads(line.replace("'", '"')) for line in output_lines if line.startswith('{')]
        for data in location_data:
            key = data['key']
            latitude = float(data['lat'])
            longitude = float(data['lon'])
            timestamp = int(data['timestamp'])

            # Calculate the distance between the current location and your home location.
            current_location = (latitude, longitude)
            distance_miles = get_distance(current_location, home_coordinates)
            print(f"Checking '{key}' - Distance from home: {distance_miles:.2f} miles")

            # Check if an alert has been sent for this item and it's more than 1 mile from home.
            if key not in item_alert_status and distance_miles > 1.0:
                send_telegram_alert(key, distance_miles)
                item_alert_status[key] = True

        print("Geofencing check completed.")

    except Exception as e:
        print("Error:", e)

    # Wait for 5 minutes before checking again.
    print("Waiting for the next geofencing check...")
    time.sleep(5 * 60)
```

Systm21, 2023-10-17:
@biemster How often do the Lenze devices advertise the key with your code?

When i have a look at it, with nrf connect, it varies from 8 to 15sec. ?!

biemster, 2023-10-17:
@Systm21 every 5 seconds:
https://github.com/biemster/st17h66_FindMy/blob/68f7b2be9ca8ec68f289e885bdb3673ee2179153/FindMy/source/FindMy.c#L260

Loic760, 2024-01-26:
Those "fake AirTags" are also ST17H66 in a nice but rather plasticky enclosure compared to real AirTags

supaeasy, 2024-03-03:
> In short my use case is: "Using (Off-Brand) AirTags on Android devices, Bonus: HomeAssistant Integration"

Just for anyone reading up on this:
Everything works very fine and ever since this project has gone from headless to macless it even runs completely on my Synology NAS. Marvellous project! Indoor tracking works quite fine with espresense-companion for anyone interested in that. 
What astonishes me the most is that there is more  functionality than with original apple devices.

closed 2024-04-22 by dchristl

GunayAnach, 2025-01-07:
> @johnbaker26222 My trackers (ST17H66) run internationally. A script runs in a VM that retrieves the position data at definable intervals and displays it on a website.
> 
> I have repeatedly observed that a position report is made during intercontinental flights. It looks like it's always in the same place. I will keep watching.
> 
> ![bkk_zrh](https://private-user-images.githubusercontent.com/119928546/263502116-fecacf7c-3eb9-4e4f-bae1-8d707cbd00e1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyMTQyODcsIm5iZiI6MTczNjIxMzk4NywicGF0aCI6Ii8xMTk5Mjg1NDYvMjYzNTAyMTE2LWZlY2FjZjdjLTNlYjktNGU0Zi1iYWUxLThkNzA3Y2JkMDBlMS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwN1QwMTM5NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kYjQ3NDk0MjU4NzY2MjBmYmMwZjBhMGRjNDE5ODJlY2Y4NTJkNzQ3NTA4ZWY1NGQ3OTZjZTAzM2RhMTcxM2VhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.l77uzu788cVqtebZAfk2lKxwnav2SWGWOSQh5bGRnhs)

This looks awesome. I found a way to extract findmy data to KMZ files, but unsure how to visualise them on a map. 
Any chance you can share the code and website? Or is it proprietary?