dchristl/macless-haystack #18: Importing my JSON file fails

johnbaker26222, 2023-06-19:
When running the python simplehttp server locally on my machine, I cannot import my JSON file. I can see my accessories after selecting the file, but when I click on import I go back to "There is nothing here yet add accessories to continue" 
Not seeing any errors in the terminal. Do I need to place my JSON file somewhere in the web directory before running the http server?

dchristl, 2023-06-19:
Hello @johnbaker26222 ,

have you pressed the import button in the upper right corner, too? 

[image]

If you go back (upper left) all changes are discarded.

johnbaker26222, 2023-06-19:
Hey thanks for the quick response. Yes clicked the import button and it takes me back with no accessories

johnbaker26222, 2023-06-19:
I also cannot create a new accessory

dchristl, 2023-06-19:
- How many keys have you generated?
- How long have you waited after the import?
- Is your fetch-location service running?
- Do you have any plugins, blocker in your browser? Can you try a different one?
- Do you see any errors in the console of your Browser (normally F12)?
- Do you have the same result at [my Github-Page](https://dchristl.github.io/headless-haystack)?
-

johnbaker26222, 2023-06-19:
> * How many keys have you generated?
6
> * How long have you waited after the import?
it immediately goes to the page with no accessories
> * Is your fetch-location service running?
john@Johns-Mac-mini webserver % Using python2
serving at port 6176

> * Do you have any plugins, blocker in your browser? Can you try a different one?
I tried chrome and firefox same same
> * Do you see any errors in the console of your Browser (normally F12)?
main.dart.js:4720 Uncaught TypeError: Cannot read properties of undefined (reading 'generateKey')
    at main.dart.js:94267:36
    at aLm.a (main.dart.js:5999:62)
    at aLm.$2 (main.dart.js:43271:14)
    at Object.G (main.dart.js:5985:10)
    at ah5.acr (main.dart.js:94279:10)
    at ah5.oC (main.dart.js:94247:21)
    at main.dart.js:94291:14
    at aLm.a (main.dart.js:5999:62)
    at aLm.$2 (main.dart.js:43271:14)
    at Object.G (main.dart.js:5985:10)
> * Do you have the same result at [my Github-Page](https://dchristl.github.io/headless-haystack)? When I usee your page it will import. But I cannot get it to fetch location data. I get this error in the console of browser:
Uncaught Error
    at Object.aW4 (main.dart.js:7570:24)
    at aby.$1 (main.dart.js:101402:53)
    at aH5.aAL (main.dart.js:46775:34)
    at aH5.O0 (main.dart.js:46777:21)
    at aDg.$0 (main.dart.js:46070:11)
    at Object.Bq (main.dart.js:6346:40)
    at aq.oQ (main.dart.js:45999:3)
    at Object.ben (main.dart.js:6446:8)
    at auI.$1 (main.dart.js:46227:9)
    at aCo.$1 (main.dart.js:50573:21)

johnbaker26222, 2023-06-19:
When using your page I seen the code: `main.dart.js:49845 Mixed Content: The page at 'https://dchristl.github.io/headless-haystack/#/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://101.108.110.114:6176/'. This request has been blocked; the content must be served over HTTPS.` 
So running chrome with the --allow-running-insecure-content fixed the error and everything worked as it should. I'm not sure how I'm going to do this on my mobile device though. Is there any way you could host an http version of your page? or implement an SSL connnection?

dchristl, 2023-06-20:
OK, I see the problem here. It is because the fetch location service is not served over https, but the github page or your local webserver is. I will have a deeper look into this next week, but it is not so easy to solve. The main problem is that you need a certificate for the fetch location service to avoid the error.

johnbaker26222, 2023-06-20:
Ok thanks any idea what is causing this error in my local instance?
Uncaught TypeError: Cannot read properties of undefined (reading 'generateKey')

johnbaker26222, 2023-06-24:
Thanks for that I do indeed have all of those files so who knows why my accessories will not import when I run a local server. 
I did find a work around for the SSL error. I generated valid certs using letsencrypt and used those with stunnel to intercept the traffic from the FindMyproxy and convert to SSL so the browser would stop complaining. This seems to have worked so I finally have a way to track on my mobile devices. 
Next up is to write a python script to send me telegram notifications when my accessories move more than a few KM.
Thanks so much for you work here!

dchristl, 2023-07-03:
FYI: 
I've added a section about the problem as last paragraph of [set up your virtual or real MAC - section](https://github.com/dchristl/headless-haystack/blob/main/OSX-KVM/README.md). This works with self signed certificates (included), too. Although your solution is a way better, but harder to set up for the most users, especially when server names changes (internal and external name).

I will close this issue now, if you have a follow up problem/ question, just create a new one. Thanks

closed 2023-07-03 by dchristl

johnbaker26222, 2023-07-05:
Awesome thanks for taking the time to fix this!! When my cert expire I’ll do the self signed method