dchristl/macless-haystack #21: Connection to Home Assistant.

Systm21, 2023-08-24:
Hey @dchristl,
do you think you can integrate some kind of connection for Home-Assistent? 
Would be really great to use haystack/findmy data in automations or something like that. 
Maybe mqtt can be a start, so home-assistent can autodiscover devices in mqtt and add it without any struggle. 

I know there are more people interested, but at the moment, nobody has the knowledge to develop a "bridge".

beckmx, 2023-08-24:
I have that connection made, but as maaaany people have said to you, that’s a security thread and abuse of the findmy network

Systm21, 2023-08-25:
Maaany people said to me? 
Don't know what you mean, writing the first time about it. In the Homeassistent community board, there are some guys talking about it, so i thought there are more interests.

dchristl, 2023-08-26:
Sorry, but this a way out of scope for this project. I also don't use HomeAssistant and therefore have no experience.

closed 2023-08-26 by dchristl

Systm21, 2023-08-26:
> I have that connection made, but as maaaany people have said to you, that’s a security thread and abuse of the findmy network

Would you share your solution with me? I wont share it with somebody else.