dchristl/macless-haystack #22: Webserver import

23tom2323, 2023-08-25:
Hello,

I'm currently testing a couple of ESP32 with the OpenHaystack firmware.
Unfortunately there is a big problem with the mail plugin in the MacApp.

I am currently retrieving the coordinates using a Python script.
python scripts/locations_cli.py --key-id XXXXXXXXXXX --private-key YYYYYYY
The output is a list with:
timestamp=datetime.datetime(2023, 8, 24, 17, 12, 49)), LocationReport(latitude=50.9559506, longitude=6.6041557, confidence=156, status=0,
This is all very cumbersome and I wanted to take a look at your project.
There are 2 import options on the web server.
1.Import Accessory and 2.Import from JSON File.

For import accessory need ID and private key (Base64)
I get it from the OpenHaystack (MacApp)
(Right click on the tracker den -> Copy ID (Base64) and Copy private Key B64)

If I add it to the web server - Import Accessory and go to Import, I get the error message: ID must be an integer value.

When importing from JSON file I'm missing the JSON ;-) where or how do I get that?

dchristl, 2023-08-26:
Hello @23tom2323 ,

> I'm currently testing a couple of ESP32 with the OpenHaystack firmware.
Unfortunately there is a big problem with the mail plugin in the MacApp.

This is one of the reasons why this project exists. ;)

> If I add it to the web server - Import Accessory and go to Import, I get the error message: ID must be an integer value.

The ID (first input) has to be a number. This should work: 

[image]

> When importing from JSON file I'm missing the JSON ;-) where or how do I get that?

The json will be generated by using my generate_keys python script and not by openhaystack.

Kind regards,
Danny

23tom2323, 2023-08-27:
Tested it with “https://dchristl.github.io/headless-haystack/”. Unfortunately, no location comes up with the error message “Could not Find location report. Try later.

If I install it locally on the Mac I get 127.0.0.1:6176

```
Directory listing for /

__pycache__/
apple_cryptography.py
FindMy_proxy.py
rename_me.pem
web/
```

dchristl, 2023-08-27:
> Tested it with “[https://dchristl.github.io/headless-haystack/”](https://dchristl.github.io/headless-haystack/). Unfortunately, no location comes up with the error message “Could not Find location report. Try later.

How is your setup?  Do you use a virtual Mac and ty to connect from the host or is everything on your real Mac? The default setting of the app is to have a virtual Mac and call the API from your host. The virtual Mac exposes port **5**6176 instead of 6176. So, checkout your settings in the app, if you are using everything on your real Mac and change the API-Port to 6176:

[image]

> If I install it locally on the Mac I get 127.0.0.1:6176

Thi is correct, the API will be called by POST. If you are using a browser, it is a GET-request, which means it shows the directory listing by default.

23tom2323, 2023-08-27:
So FindMy_proxy.py is running locally (python3 FindMy_proxy.py) and additionally need the https://dchristl.github.io/headless-haystack/ page where I specify my local server with port for communication. Aha :-)

I thought it would run locally and https://dchristl.github.io/headless-haystack/
would only be a demo / example of the local version.

It worked with 1 tag that was entered as a test twice.
I deleted the duplicate tag and entered my other tags (4 pieces). Since then he doesn't want to talk to the local FindMy_proxy anymore.
Have already restarted the FindMy_proxy, checked the port again on both sides and also changed the port on both sides.
Unfortunately, both of them don't want to talk to each other anymore

dchristl, 2023-08-27:
> So FindMy_proxy.py is running locally (python3 FindMy_proxy.py) and additionally need the https://dchristl.github.io/headless-haystack/ page where I specify my local server with port for communication. Aha :-)
> I thought it would run locally and https://dchristl.github.io/headless-haystack/
would only be a demo / example of the local version.

Have you considered reading the project description and the detailed guide? The main purpose of the project is to operate OpenHaystack completely without a Mac (for users without a Mac). Unfortunately, this cannot be achieved 100% as the Apple Keystore is required for authentication. Therefore, the API is running there (FindMyProxy-script on port 6176) and can be accessed by three possible clients: an Android app or my website hosted on Github, or a website that you host yourself.

> Have already restarted the FindMy_proxy, checked the port again on both sides and also changed the port on both sides.
Unfortunately, both of them don't want to talk to each other anymore

Do you see the calls in the log where FindMyProxy is running? The requests are logged something like this: "Getting with Post ..." Can you still perform the previous GET request in the browser? Does the directory listing show up there? Are you still calling the API via localhost? Is there an error in the browser console? If yes, what error message is displayed?