dchristl/macless-haystack #23: Map not showing.

Systm21, 2023-08-26:
Is there somebody else, that have problems with the map, when opening web or android Application?

[image]

I always have to zoom out a bit and then it works nicely. Sometimes the map didn't come back, when zooming in to much, so you have to restart the browser/app. 

Tried this several times with the "original" openhaystack app, there are no mistakes like that.

dchristl, 2023-08-27:
Yes, unfortunately, I have the same issue with the web app. The Android app works without any problems. There seems to be an issue with the Flutter implementation of the map, but I haven't had a chance to look into it further. By the way, is this also an "original" OpenHaystack implementation? My modifications only relate to the called API.

Systm21, 2023-08-27:
Yes it's also haystack, my description belongs to the seperation of the projects. 

It feels like the map problem in the web app is much more frequent and the crashes that need a reload of the page are much more frequent, but the Android app also has one or the other problem with the map. The map does not appear every time you start it, it only loads when you zoom out a bit.

dchristl, 2023-09-04:
I've found a work-a-round at the [issue tracker of flutter_map ](https://github.com/fleaflet/flutter_map/issues/1563), that is implemented in the latest version. It works well for me now on web and Android.

Please try it out and give feedback. Thank you.

Systm21, 2023-09-05:
Works fine now, will have an eye on that in the next days. 

Another question to the map:
Why is the app /web only using 20 reports per tracking device? 
The history makes no sense with 20 reports, because its only 1-2 days in rural area. In the inner-city certainly a lot less.

dchristl, 2023-09-08:
Great, that it works now. 

> Why is the app /web only using 20 reports per tracking device?
The history makes no sense with 20 reports, because its only 1-2 days in rural area. In the inner-city certainly a lot less.

I really can't answer this ;) I have adopted the openHaystack project almost 1:1 and only made a few changes regarding the API implementation. Everything else, including the numerous errors, remains as it is. Unfortunately, I lack the time to fix everything, but in my view, it is sufficient for basic functionality. The goal of the project is also to utilize OpenHaystack functionality completely (or with minimal Mac-specific components). You are welcome to contribute to the project (e.g., through pull requests).

Systm21, 2023-09-09:
...but the App from the Openhaystack Project works as expected. Their app also not showing how much reports are used, so there were some changes. 
The other app shows a complete history of the last 7 days.

dchristl, 2023-09-09:
The have two apps. A native one (written in swift), that they have released and a [second one](https://github.com/seemoo-lab/openhaystack/tree/main/openhaystack-mobile) (written with flutter), that is the base of this project. The native one runs only on Mac, the other theoretically one on any device, that they don't release.

Systm21, 2023-09-09:
I not really want to compare cherrys with eggs so everytime i write about the app, its the android/web app written in flutter..
The native app is a complete different thing, i know.

dchristl, 2023-09-10:
Ok, now I see what you mean, but the changes are slightly there. I've added the number of history reports and the decryption is async now, because of the long waiting time. The rest is like before. I will have a look at it when I have time.

closed 2023-10-09 by dchristl