dchristl/macless-haystack #26: dd: no value specified for seek

0xPetra, 2023-09-13:
```
dd: no value specified for seek
```

I'm getting this error while trying to run this command:

```xxd -p -c 100000 PREFIX_keyfile | xxd -r -p | dd of=nrf52_firmware.bin skip=1 bs=1 seek=$(grep -oba OFFLINEFINDINGPUBLICKEYHERE! nrf52_firmware.bin | cut -d ':' -f 1) conv=notrunc```

I also tried running grep and got nothing as a response:

```
grep -oba OFFLINEFINDINGPUBLICKEYHERE! nrf52_firmware.bin | cut -d ':' -f 1
```

I compiled the `.bin` files again. Something that I need clarification on is how this command would look for a string in a binary file.

I tried replacing `OFFLINEFINDINGPUBLICKEYHERE! ` with the actual PUB KEY (which doesn't make sense), but I still did it. 

What could be happening? 
Appreciate the repo!

0xPetra, 2023-09-14:
I'm using `zsh`, so it should work.

Sorry to ask such a simple question, but where should I download it from? 

Besides compiling my own version I also downloaded from here: 
```https://github.com/acalatrava/openhaystack-firmware/tree/main/apps/openhaystack-alternative/compiled```

Thanks!
Cheers,
Petra

dchristl, 2023-09-14:
> I'm using zsh, so it should work.

Do you use it on Linux or Mac? It seems Mac has an issue with grep. Please try to set the LC_CTYPE env variable before the command, i.e.:

`LC_CTYPE=C grep -oba OFFLINEFINDINGPUBLICKEYHERE! nrf52_firmware.bin | cut -d ':' -f 1`

resp.

`LC_CTYPE=C xxd -p -c 100000 PREFIX_keyfile | xxd -r -p | dd of=nrf52_firmware.bin skip=1 bs=1 seek=$(grep -oba OFFLINEFINDINGPUBLICKEYHERE! nrf52_firmware.bin | cut -d ':' -f 1) conv=notrunc`

> Sorry to ask such a simple question, but where should I download it from?

You should use the version from this project at the [release page](https://github.com/dchristl/headless-haystack/releases/tag/v1.5.0)

0xPetra, 2023-09-14:
Oh, I didn't realize about releases. After making sure everything works, I can update the Readme with some notes.

Indeed I'm using a Mac. `LC_CTYPE=C` does work for grep, printing `155224`, which is the expected value as you mentioned. But the second command still throws an error with the new .bin:

```
dd: no value specified for seek
```

dchristl, 2023-09-14:
I've updated the README, to execute the command on non -GNU shells. The changes should work for Linux and Mac.

dchristl, 2023-09-18:
> arm-none-eabi-objcopy -I binary -O ihex nrf52_firmware.bin nrf52_firmware.hex 

I'm not familar with the programmer app, you are using, but this shouldn't be needed normally. I flash the firmware with opeonocd, that works really well. The commands and the openocd.cfg file is in the docs. Maybe you have to change the last line openocd.cfg from

`source [find target/nrf51.cfg]`  to
`source [find target/nrf52.cfg]`, but I don't think this is really needed.

 Can you try it with openocd? 

> I thought maybe SoftDevice was missing and tried to add it separately. But it overlaps with the firmware, so I guess it was indeed included while compiling.

You are right, the softdevice is already included.

closed 2024-01-17 by dchristl