dchristl/macless-haystack #27: Problem setting up dockerized MacOS in WSL2

careyer, 2023-09-19:
Hi there,

i managed to install a dockerized MacOS running Windows 10 + Docker-Desktop + WSL2.
However I struggle with this step of the HowTo (https://github.com/dchristl/headless-haystack/blob/main/OSX-KVM/README.md) 

```
sudo su
cd /var/lib/docker/containers/89f71e516b9f_________
```  

There simply ain't no folder `/var/lib/docker/containers` ... 
Somehow Docker-Desktop in conjunction with WSL2 stores the containers somewhere else but I cannot find the location for the heck of it. Any advice?

Thanks in advance!

careyer, 2023-09-19:
Seems like at least Win11 is needed to get OSX-KVM to run properly. Win10 does not support nested virtualization. 
This information should be added to the HowTo.

dchristl, 2023-09-22:
Hello @careyer,

thanks for pointing this out, but I'm not sure this is correct. I can't try it out myself, because I have no native Windows host. But I think nested virtualization is also supported on Win10, but not on any CPU. Also it should work without virtualization, too. It will be incredible slow, but after setting up the machine (Login with AppleID) and running headless this shouldn't be an issue anymore. 

Kind regards,
Danny

biemster, 2023-10-19:
@dchristl Some really interesting stuff is happening on this front in https://github.com/seemoo-lab/openhaystack/issues/63#issuecomment-1769216337, we might soon be able to get rid of running macOS in a VM entirely!

dchristl, 2023-10-19:
@biemster thank you, I've seen it already and it sounds amazing. Unfortunately I don't have so much free time currently, but I will have a deeper look as soon as I can. Hope this will be running on an raspberry pi. That would be great.

dchristl, 2024-01-17:
This issue is not valid anymore with the latest release.