dchristl/macless-haystack #28: AHow many days back?

Systm21, 2023-10-09:
First of all, nice improvement. I think i like the big dot, where several reports habe been received within a time period. Is it dynamic? Is it growing if there are more locations very near, or is it just a big dot to show, here are  more than xy location reports in the last hours?

But there is a bug, the slider "How many days back?" does nothing. a place where ive been on friday is showing, even if i set it to 1 (days back).

dchristl, 2023-10-09:
> I think i like the big dot, where several reports habe been received within a time period. Is it dynamic? Is it growing if there are more locations very near, or is it just a big dot to show, here are more than xy location reports in the last hours?

It is a little bit dynamic. There are 4 steps and every step is 6 (6,12, 18, 24) hours. The default dot is below 6 hours until the largest dot over 24 hours.  You can see the timespan on hover/ click over the dot. 

> But there is a bug, the slider "How many days back?" does nothing. a place where ive been on friday is showing, even if i set it to 1 (days back).

I will have a look into this.

Systm21, 2023-10-09:
I keep an eye on that, with the small database i have available at the moment, it works.

closed 2023-10-09 by Systm21