dchristl/macless-haystack #31: iOS 17 devices not sending reports?

shiprec, 2023-12-09:
Hello we have noticed that there are no location reports for our tags when they are near IOS17 devices.  Everything works fine for IOS16 but not 17.  Anyone else seeing this and if so any ideas on where the reports are now going?

dchristl, 2023-12-10:
Hello, sorry I can't really help you with that, because this project is not about IOS. If Apple found a way to distinguish between real Apple devices and the fake ones all projects like (openhaystack, FindMy, ... ) will be dead soon. :(

shiprec, 2023-12-10:
Thank you for the response.  I was wondering if maybe the reports were going to a different location rather then the tags themselves being ignored.

closed 2023-12-14 by shiprec

careyer, 2024-01-28:
Hey @dchristl,

I am afraid to confirm this. When the ESP32 tag is near my two iOS17 devices not location reports are created. I tried for several days. Only when my old ipad (iOS 16) and/or my old iPhone6 (iOS 16) are arround location reports are rolling in.

dchristl, 2024-01-29:
There is already a discussion about it going on [here](https://github.com/biemster/FindMy/issues/40) . 
I'm monitoring this and will react accordingly. Currently, it looks like Apple messed something up.