dchristl/macless-haystack #32: macless docker not runnig in ubuntu

Systm21, 2024-01-02:
First of all, the Anisette Server sets up without any problem, but the API-Server does not come up. 

When i take the Scripts and dependencies, i can set up an endpoint, without docker. 
When i run it with docker, the only log entry is: `exec /usr/local/bin/python: exec format error` and the server didn't come up.

dchristl, 2024-01-03:
Hello Systm21,

 

I haven't really finished implementing Docker yet, but in principle, it should work. Did you build the image yourself or did you pull it from Docker Hub? How did you start the image, as specified in the instructions? Could you try pulling the image completely anew? I've been experimenting with intermediate states, and some of them were broken.

 

Kind Regards,

Danny

Systm21, 2024-01-03:
Hey Danny,
cloned the macless repo and fired up the docker-compose.yml in the endpoint dir.

dchristl, 2024-01-04:
Please ignore the compose file. I haven't actively maintained that and suggest referring to the instructions at the [README] (https://github.com/dchristl/macless-haystack/blob/macless/README.md). There's no need to check the repository out. In my opinion, it's quite convenient now, as it only involves 3 CLI commands.

I appreciate it if someone could try it out and provide feedback, as I'm relatively new to Docker-related matters

dchristl, 2024-01-17:
This issue is not valid anymore with the latest release.

closed 2024-01-17 by dchristl