dchristl/macless-haystack #37: generate_keys.py fails to run

goliath29, 2024-01-21:
I have installed all dependencies, though when attempting to run the generate_keys.py file it reports back the following error.

  File "C:\Users\XXX\XXX\esp32-firmware\generate_keys.py", line 6, in <module>
    from cryptography.hazmat.primitives.asymmetric import ec
ModuleNotFoundError: No module named 'cryptography'

I am unsure if there is a dependency I have missed, as I have gone through the documentation on this repo and the others linked to no avail. 

Any assistance with this error message would be greatly appreciated.

dchristl, 2024-01-21:
Hello @goliath29 ,

you have to install the module with:

`pip install cryptography` or
`pip3 install cryptography`

I will soon clearify the documentation at this point.

Kind Regards,
Danny

goliath29, 2024-01-22:
Amazing, it works now!

Thank you for the help

closed 2024-01-22 by goliath29