dchristl/macless-haystack #39: Error when trying to refresh / get location from server

Loic760, 2024-01-21:
Hey,

with the version 2.0 and GitHub website I am encountering the following error when refreshing the position "Could not find location reports. Try again later. Error: ClientException: XMLHttpRequest error., uri=http://localhost:6176"

I can see the requests being received by macless on docker so I am not sure about where is the problem

dchristl, 2024-01-21:
Hi @Loic760 ,

please check out the [FAQ](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md). The command is:

`docker logs -f macless-haystack`

Regards,
Danny

Systm21, 2024-01-21:
Seems like you dont have changed the url on the Website, when u use the Github Page from @dchristl.

Loic760, 2024-01-22:
works well with https://macless-haystack.orb.local (I am using docker through OrbStack) but not working with http://localhost:6176 or http://127.0.0.1:6176 despite the website running same as with the OrbStack local url. Does it relies on a "domain name" or url without port to work ?

dchristl, 2024-01-22:
I'm not familiar with the exact workings of orbstack, so I don't understand how you've set it up in detail. I'm just surprised that the orbstack URL is HTTPS, while your localhost URL is HTTP on Port 6176. They can't be the same system. If it's the same system and you have SSL activated, the URL should be https://localhost:6176/

closed 2024-03-18 by dchristl