dchristl/macless-haystack #41: 📦 Pack everything in a container...

Systm21, 2024-01-22:
First of all, really brilliant work you are doing here.

Is it possible to pack the web client into its own container with auto-update?

dchristl, 2024-01-23:
I tend to and, honestly, am more inclined to completely remove the client from the releases. From my perspective, there is simply no reason to want to self-host the web frontend. The GitHub page is always up-to-date, secure, and highly available. Achieving that level of reliability is hardly possible with self-hosting. 

If you have a valid reason, feel free to let me know, and I may reconsider my decision.

Additionally, anyone could always check out the repository and host the server themselves. Updates can then be obtained through Git.

Systm21, 2024-01-23:
I don't think I'm the one who has to give any reasons here. If you don't want to hold on to the open source idea anymore (which is what it sounds like when you say you don't want to publish this in the future), then go ahead. Is this change of heart just because I asked for a Docker version? 
If so, a simple no would have done the trick. 

Why am I hosting:
-Host a server myself (there are plenty of providers) 
-Nextcloud myself (there are enough providers) 
....

I could go on like this indefinitely, but the answer is always the same: because it feels better. 
...and don't tell me "I don't save any data, it's all in the cache". That may be true, but self-hosting is trending, just take a look at reddit.

dchristl, 2024-01-24:
OK, there seem to be some misunderstandings in my statement and uncertainties about how the application works. Perhaps I expressed myself unclearly, possibly because I am not a native speaker. First of all, the project is and will remain open source. What I am considering not releasing anymore is only the webapplication.zip, the rest remains as is. Open source, by the way, does not dictate whether I release a compiled version, only that the source code is open, and it is. I had this consideration even before your request for your Docker container.

The background behind this is that I can invest much less time in this project than is necessary. This will probably decrease further due to personal matters. I need to prioritize this time somehow and would prefer to invest it in meaningful advancements rather than things that are not inherently necessary. The web application will still continue to function on GitHub, but maintaining it, as well as a potential Docker container, involves effort.

 

I have been hosting my own servers for many years, and the reasons for doing so are well understood. But only if it makes sense. The most common reason (also mine) is the sovereignty of data. Data should not be handed over to corporations that use it for analysis and advertising purposes. However, in the case of the web application via GitHub, this is completely irrelevant, as no data is sent to GitHub. The application (HTML, CSS, JS) is delivered, and then no requests are made to GitHub anymore. They know nothing about the location, which keys you have stored, or any settings. Everything remains local or is transmitted to the endpoint. So, there are no advantages to self-hosting. If you think about it correctly, you would actually have to host the map provider (OpenStreetMap) yourself, as most and easily private data (the map sections) are transferred here. Do you want to set up a Nominatim additionally? However, I would always host the endpoint myself, as, for example, the (public) keys are involved here, which is why it remains as a container.

 

My question about the reasons was meant seriously. Maybe I'm overlooking something, and there are use cases where self-hosting the web application could make sense, I just can't think of any.

Kind Regards,

Danny

Systm21, 2024-01-24:
The webapplication.zip can stay out, everyone should be able to compile it quickly with Flutter.  I also always do this with Flutter, I find the process somehow faster.

closed 2024-01-24 by Systm21

Systm21, 2024-03-27:
@mrx23dot 
Look at this: https://github.com/Chapoly1305/FindMy