dchristl/macless-haystack #42: How to esacpe special characters in Apple ID Password?

careyer, 2024-01-25:
Hey there!

is there a way to Escape "special characters" in the password entry when creating the macless-haystack docker container in interactive mode?

My password is constructed like this: " %5something5% " . I think it has problems the with the  % character.
I always get this error message

```
2024-01-20 11:41:27,605 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-20 11:41:27,607 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-20 11:41:27,609 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-20 11:41:27,611 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-20 11:41:28,345 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1269
2024-01-20 11:41:28,347 - WARNING - This implementation only supports s2k. Server returned s2k_fo
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 124, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 76, in registerDevice
    getAuth(regenerate=True, second_factor='trusted_device' 'sms')
  File "/app/endpoint/register/apple_cryptography.py", line 49, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 41, in icloud_login_mobileme
    pet = g["t"]["com.apple.gs.idms.pet"]["token"]
          ~^^^^^
TypeError: 'NoneType' object is not subscriptable
```

careyer, 2024-01-25:
Okay... created a new account with a simple password and 2FA.... now I get a step further but...

```
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-01-25 23:25:17,865 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-01-25 23:25:17,866 - INFO - No auth-token found.
2024-01-25 23:25:17,866 - INFO - Trying to register new device.
Apple ID: tomyb@gmx.net
Password:
2024-01-25 23:25:30,338 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:25:30,348 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:25:30,356 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:25:30,362 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:25:31,288 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1270
2024-01-25 23:25:33,547 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:25:33,549 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:25:33,553 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:25:33,557 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:25:34,544 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 18119
2024-01-25 23:25:34,640 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:25:34,642 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:25:34,646 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:25:34,650 - DEBUG - Starting new HTTPS connection (1): setup.icloud.com:443
2024-01-25 23:25:35,112 - DEBUG - https://setup.icloud.com:443 "POST /setup/iosbuddy/loginDelegates HTTP/1.1" 200 303
2024-01-25 23:25:35,117 - DEBUG - Answer from icloud login
2024-01-25 23:25:35,118 - DEBUG - {'dsid': '13000021', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'Account limit reached.'}}, 'status': 0}
2024-01-25 23:25:35,119 - ERROR - Invalid status: 1
2024-01-25 23:25:35,119 - ERROR - Error message: Account limit reached.
2024-01-25 23:25:35,119 - ERROR - Unable to proceed, program will be terminated.
pi@Pi4B-macless-haystack:~/IOTstack $ docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-01-25 23:27:57,415 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-01-25 23:27:57,416 - INFO - No auth-token found.
2024-01-25 23:27:57,417 - INFO - Trying to register new device.
Apple ID: tomyb@gmx.net
Password:
2024-01-25 23:28:11,301 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:28:11,311 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:28:11,321 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:28:11,330 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:12,219 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1270
2024-01-25 23:28:14,518 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:28:14,520 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:28:14,524 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:28:14,528 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:15,467 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 5068
2024-01-25 23:28:15,563 - INFO - 2FA required, requesting code
2024-01-25 23:28:15,572 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:28:15,575 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:28:15,580 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:28:15,583 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:16,451 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1270
2024-01-25 23:28:18,805 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:28:18,807 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:28:18,812 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:28:18,815 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:19,832 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 5068
2024-01-25 23:28:19,838 - INFO - 2FA required, requesting code
2024-01-25 23:28:19,838 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-01-25 23:28:19,840 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-01-25 23:28:19,845 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-01-25 23:28:19,849 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:21,232 - DEBUG - https://gsa.apple.com:443 "PUT /auth/verify/phone/ HTTP/1.1" 200 None
Enter 2FA code: 855833
2024-01-25 23:28:37,874 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-01-25 23:28:39,815 - DEBUG - https://gsa.apple.com:443 "POST /auth/verify/phone/securitycode HTTP/1.1" 200 None
2024-01-25 23:28:39,817 - INFO - 2FA successful
2024-01-25 23:28:39,834 - DEBUG - Querying http://anisette:6969 for an anisette server
[... cut ...]
2024-01-25 23:28:45,754 - DEBUG - https://setup.icloud.com:443 "POST /setup/iosbuddy/loginDelegates HTTP/1.1" 200 303
2024-01-25 23:28:45,758 - DEBUG - Answer from icloud login
2024-01-25 23:28:45,759 - DEBUG - {'dsid': '13000021', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'Account limit reached.'}}, 'status': 0}
2024-01-25 23:28:45,759 - ERROR - Invalid status: 1
2024-01-25 23:28:45,760 - ERROR - Error message: Account limit reached.
2024-01-25 23:28:45,761 - ERROR - Unable to proceed, program will be terminated.
```

careyer, 2024-01-26:
**Update:** Okay got it working with my new AppleID Acccount 
However:
- adding a paypemt method to the account was not enough
- I needed to link a real Apple Device to the account (old apple iPhone6) first

[image]

After addinng the other device all worked fine!
However the problem with entering passwords containing special characters (e.g. % remains)

dchristl, 2024-01-26:
Hello @careyer,

the problem with the error `Account limit reached.` is a known problem. Glad you find a solution. I will add this to the FAQ soon.

I will have a deeper look at your problem with the password.

Kind Regards,
Danny

dchristl, 2024-01-27:
I unfortunately cannot reproduce your issue with passwords and special characters. I have now created a password with all possible combinations, and all have worked (`" %5Something5% "`, `%5Something5%`,` %5something5% `(leading/trailing spaces not accepted by Apple)). 

I suspect that your terminal might be transmitting incorrect characters. What do you use to start Docker (which OS, which terminal)? You could try recreating your container by removing it and then creating it again with:

```
docker rm -f macless-haystack
docker run -i --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack
```

The only difference is the absence of the -t parameter. This should result in your password being echoed to the console.

careyer, 2024-01-28:
I will try but have to setup a 2nd test instance first (don't want to kill the container - now that I got it working after trying for several hours).

I am using PiOS 64bit, my CLI is bash and I am connecting via ssh (using Putty). Since my linux root password also includes the "%" character I think the terminal is safe to say to transmit the character correctly. Other than this character nothing is different to the 2nd Apple account that I created.

Nevermind maybe something else was giving problems!

closed 2024-04-22 by dchristl