dchristl/macless-haystack #44: esp firmware

ju5t3nc4s3, 2024-01-30:
i like to compile my own firmware for a newer board ESP32-C2 .were to start cant find that this board is compatable with IDE.  i have these nano boards from aliexspress esp8684 wroom 07 . im gona itergrate them into usb cables

dchristl, 2024-01-31:
Hello,

I don't understand everything what you are trying to achieve, but your error seems, that the folder structure ist not correct. This is because the project is in "platformio"-style and not the default for the esp32 toolchain. You can have a look at my [other project](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md) the instructions are working here, too. You just have to add your board to platform.ini to compile for ESP32 C2. You can find an example how this is will be done [here](https://github.com/dchristl/esp32_nat_router_extended/blob/master/platformio.ini) 

Regards,
Danny

ju5t3nc4s3, 2024-01-31:
i cant get the board manager to show up for this esp32-c2/esp8684 in platformio or in arduino ide.  this is why i tryed to buils the firmware with exspressif ide.py  it shows up with 5.1

ju5t3nc4s3, 2024-02-11:
is it posable to get this into 2mb flash its showing i need  (.1mb)

dchristl, 2024-02-11:
Sorry, I don't understand what you are trying to say. The firmware is below 1 MB.

ju5t3nc4s3, 2024-02-12:
finaly got it running,   when im done this will be the worlds smallest gps tracker, it will be imbeded into a usb charge cable and very undetectable
[image]
[image]

supaeasy, 2024-03-04:
Interesting but I don't understand your use case.

closed 2024-04-22 by dchristl