dchristl/macless-haystack #45: App not having location but in docker DEBUG log I get some data

TepMaster, 2024-02-08:
As I stated in the title via the mobile app i get no location(Unknown) but in docker I get this JSON object ```{
   "results":[
      {
         "datePublished":1707413215833,
         "payload":"K3VEwQMEoenmdFExNMceJBWtQEqmwMduPxlQv9CECEdMc27L0eRpQqDQbAx9amsWQs4NOM5uU7/80OvxU5Uw8QRpTGmg4raTptg0hWRtCkJRtMOKcVUW1Q==",
         "description":"found",
         "id":"6hAIahRUzZGDPl8KdE553DH579yOypu5yFCPKS2lXME=",
         "statusCode":0
      },
      {
         "datePublished":1707413215833,
         "payload":"K3VE1wMEUL7YCWS+mySFftsDmLO7WYrG1SshC7VDRRlX5NJMpSVB3x+r245Q119TbSuIaKGsqkDL/Dp7+Y9VAxuIH5LJOhuLGI8UFEu/fgwZPaKpxr5fYA==",
         "description":"found",
         "id":"6hAIahRUzZGDPl8KdE553DH579yOypu5yFCPKS2lXME=",
         "statusCode":0
      }
   ],
   "statusCode":"200"
}```

TepMaster, 2024-02-09:
https://imgur.com/a/iC85Wan
I tried ```adb logcat -s de.dchristl.headlesshaystack``` but got only 
```
--------- beginning of crash
--------- beginning of system
--------- beginning of main
```

If you want you can contact me on discord:marcuradu2

Thanks for your help.

dchristl, 2024-02-10:
Ok, everything seems fine for now. Usually, an error would occur if the connection isn't working. Can you provide more details about your setup? Is the container hosted on the web/cloud, or is everything local? Are you using SSL and/or BasicAuth? Have you imported the JSON or otherwise imported your accessory? The first log appears at the moment when you request with the Android app, correct? Alternatively, could you try using the [web version](https://github.com/dchristl/macless-haystack)?

TepMaster, 2024-02-11:
I host everything locally via a docker container. I am not using SSL just the default setup. The web page works but the android app does not.

dchristl, 2024-02-11:
The problem here is that the application has only one code base, because it is developed with flutter. There is no special code for Android nor web-version.  The only thing you can try, is to fully uninstall the app and install again.

closed 2024-04-22 by dchristl