dchristl/macless-haystack #46: 2FA not workin

maximushugus, 2024-02-18:
I have an apple account with 2FA enabled made for this project. I was already using openhaystack on a MacOS VM.
I was able to set up the docker containers.
But after lanching the docker with `docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack` and login with my appleID and password, it keeps asking for 2FA.
I don't receive SMS when it asks for 2FA so I launched my MacOS VM and on settings I clicked on "verification code", so I get a 6 digit code. But putting this one doesn't do the trick and it keeps asking for 2FA.

I tried connecting to icloud.com, and requesting my 2FA via SMS, and not putting this code in my browser for icloud.com but instead putting when macless-haystack asks but the same problem, it keep asking for 2FA.

Does someone know why ?

maximushugus, 2024-02-18:
Ok, so here is what I did : 

1. I lauched anisette on docker with `docker run -d --restart always --name anisette -p 6969:6969 --volume anisette-v3_data:/home/Alcoholic/.config/anisette-v3/lib/ --network mh-network dadoum/anisette-v3-server`. I verified it seems to be working because If I do `curl localhost:6969` I get an answer.
2. Then I `git clone https://github.com/biemster/FindMy` and `cd FindMy`
3. To make it work I had to install `pip install cryptography` and `pip install pbkdf2` and `pip install srp` and `pip install pycryptodome`
4. I run `./request_reports.py`

Here is my output : 

> ubuntu@vm:~/FindMy$ ./request_reports.py
Apple ID: myappleid@mail.com
Password: 
pyprovision is not installed, querying http://localhost:6969 for an anisette server
pyprovision is not installed, querying http://localhost:6969 for an anisette server
2FA required, requesting code
pyprovision is not installed, querying http://localhost:6969 for an anisette server
Enter 2FA code:

At this point I do not receive SMS nor I have a prompr on my MacOS VM for a verification code as when I try to connect to icloud.com for exemple.
I tried to put the 2FA code I obtain by manually clicking on "obtain a verification code" on MacOS VM, but it does'nt work and the prompt above starts again
I also tried to go to icloud.com and ask for an SMS 2FA, not using it on icloud.com but instead putting it on the program, but the same result.
If I just press enter, leaving the 2FA, the same result.

dchristl, 2024-02-19:
Ok, then it seems to be more of an issue with the account rather than with the code from the project. Can you possibly create an additional Apple account (which can also use the same phone number) and try again?

maximushugus, 2024-02-21:
I tried but it didn't work.
Is it normal that I see nothing in anisette logs when I'm tring to log in ? 

> docker logs anisette -f
                     app INFO  2024-02-21T12:05:41.773 anisette-v3-server v2.1.0
                     app INFO  2024-02-21T12:05:41.796 Creating machine... 
                     app INFO  2024-02-21T12:05:41.798 Machine creation done!
                     app INFO  2024-02-21T12:05:41.798 Machine requires provisioning... 
                     app INFO  2024-02-21T12:05:43.222 Provisioning done!
[main(----) INF] Listening for requests on http://0.0.0.0:6969/

supaeasy, 2024-03-04:
> > [main(----) INF] Listening for requests on http://0.0.0.0:6969/

The IP looks wrong to me. Are you sure you setup mh-network correctly?

maximushugus, 2024-03-04:
Anisette seems to be working properly because if I do : 
`curl http://localhost:6969` I get this JSON (modified) : 

> {"X-Apple-I-Client-Time":"2024-03-04T17:42:21Z","X-Apple-I-MD":"AAAABQAXXXXXXZQJt/q2Pt1YMw7dcyqV/7AAAABA==","X-Apple-I-MD-LU":"5011D56E92AFD6A880XXXXXXXBC697D23C45985E9A1987F50B6D0CC8D7ADB9","X-Apple-I-MD-M":"z6xuBAi6XXXXXXXqJ+f3We0gJUoXb+jrbDQhkP0HtlvAd0qV87nyf+fVdZCm1aTu3/qy+Be7BBgHyS","X-Apple-I-MD-RINFO":"17996176","X-Apple-I-SRL-NO":"0","X-Apple-I-TimeZone":"UTC","X-Apple-Locale":"en_US","X-MMe-Client-Info":"<MacBookPro13,2> <macOS;13.1;22C65> <com.apple.AuthKit/1 (com.apple.dt.Xcode/3594.4.19)>","X-Mme-Device-Id":"AAXXXXXXA-773B-4AFC-866F-948E97F875FA

maximushugus, 2024-03-04:
When lauching macless-haystack I see a strange behavior, maybe this is related : 

> remote: Enumerating objects: 88, done.
remote: Counting objects: 100% (88/88), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 75 (delta 37), reused 56 (delta 25), pack-reused 0
Unpacking objects: 100% (75/75), 3.62 MiB | 18.26 MiB/s, done.
From https://github.com/dchristl/macless-haystack
  branch            main       -> FETCH_HEAD
   32ab133..e2ad25c  main       -> origin/main
2024-03-04 17:46:54,267 - INFO - No auth-token found.
2024-03-04 17:46:54,268 - INFO - Trying to register new device.
Apple ID: apple@mpalandre.fr
Password: 
2024-03-04 17:47:20,772 - INFO - 2FA required, requesting code
2024-03-04 17:47:22,892 - INFO - 2FA required, requesting code
Enter 2FA code: 326094
2024-03-04 17:48:02,547 - INFO - 2FA successful
2024-03-04 17:48:04,620 - INFO - 2FA required, requesting code
Enter 2FA code:

Here is what is strange : 

1. As you can see there are 2 lines saying its requesting 2FA. Maybe this is why even if I enter the 2FA, it's still asking for the 2nd 2FA ?
2. Even if I enter a random 2FA I get at least one line saying `2FA successful`

dchristl, 2024-03-05:
The output is very strange and each line should be there only once.
It seems like the server in the container is starting twice. Have you tried resetting everything as I suggested before? Do you have an auth.json file in the data folder (usually /var/lib/docker/volumes/mh_data/_data)?
Which operating system are you using as the host? Do you have another computer to try it out there?

maximushugus, 2024-03-05:
Yes it's strange, because as you can see each line appears only once until I give my password. Then you can see two lines for 2FA..
I tried resetting everything but it did'nt change anyting.
I'm testing this on an aarch64 plateform (my test server is on Oracle Free Tier), maybe this is causing the issue. I will try to find the version of the container, maybe this isn't the latest version for aarch64 compared to x86. But I'm almost certain I tried it on my VPS x86 with the exact same result.

dchristl, 2024-03-05:
Maybe the architecture is the issue, although there is no reason for it. I also have my endpoint running on an Oracle Free Tier, but with x86.

Output of `uname -a:`
`Linux headless-haystack 5.15.0-1052-oracle #58-Ubuntu SMP Tue Feb 13 19:43:43 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
`

It works without problems since weeks. It is really hard to help you here. Have you can tried to change your terminal application for the ssh-session?

YupengLai4, 2024-03-09:
Hi, @dchrist. I also encountered the same problem. After submitting 2FA it succeeded, but then kept asking for a new 2FA auth.
[image]
I'm using ubuntu on x86, it's a home service. Should I open 6969 or 6176 port?

a-camacho, 2024-03-12:
Same problem

coopeeo, 2024-03-14:
same issue

dchristl, 2024-03-18:
Hello @YupengLai4 ,

> I'm using ubuntu on x86, it's a home service. Should I open 6969 or 6176 port?

There is no need to open any port. Your output looks like it works in general.

Could you try to register your device with [Biemsters version](https://github.com/biemster/FindMy) . If this will work, I can go deeper or you can transfer the auth.json to macless-haystack.

YupengLai4, 2024-03-19:
> Could you try to register your device with [Biemsters version](https://github.com/biemster/FindMy) . If this will work, I can go deeper or you can transfer the auth.json to macless-haystack.

Thanks for your response! I tried both macless-haystack and the Biemster one but unfortunately the issue persisted :(

dchristl, 2024-03-20:
Then I think it is a problem with your account (Apple-ID). Maybe you can create a new one and try again. Some accounts work while others don't, but nobody has really figured out why that is yet.

trueVinton, 2024-03-25:
I had the same issue, as @dchristl mentioned the culprit was in the Apple ID account. I didn't get a 2FA SMS because the Apple ID was using an outdated phone number.
To fix:
1. Go to icloud.com and log in with your AppleId.
2. click your profile picture > Manage AppleId
3. click Account Security > Enable 2FA and make sure the trusted phone number is correct.

a-camacho, 2024-03-27:
Hi guys,

Still does not work for me.
I'm using in on a Macbook Pro 2019, with Mac OS 14.3.1 Sonoma. Should it work ?

When asking for 2FA, I receive no message or device alert.
But I can generate manually a 2FA code from my iCloud settings.

What is weird, is that any code that I enter (correct or incorrect one, like 0000) the system always says "2FA successful" and then start procedure again asking me Apple ID again.

Do you have any clue what I'm doing wrong ?
Should I even be able to run it correctly ?

Thanks a lot

fachinformatiker, 2024-03-29:
I get the messages multiple times too

> ![Screenshot for 2FA required](https://private-user-images.githubusercontent.com/57296300/311406527-73cc7270-7ea8-4b53-86a9-77fa9c896687.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE3NDE3OTAsIm5iZiI6MTcxMTc0MTQ5MCwicGF0aCI6Ii81NzI5NjMwMC8zMTE0MDY1MjctNzNjYzcyNzAtN2VhOC00YjUzLTg2YTktNzdmYTljODk2Njg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI5VDE5NDQ1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk5OGU5MjFkYjViYzQyOGRmNTY0NGRlZDQ3YWMwNmE3ZjUyYjMyYjZjODAxMTRmYjIxODJlODVhZGE4YzMzNDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MEZbqx4zK0AMnLY5uEB46z5g2tOoKVMFbHWpD7e-Aes)

dchristl, 2024-03-30:
I would like to help, but I cannot reproduce the double output issue. I have tried it on x86 (Linux, various derivatives) and on Armv8, and I always receive only one prompt for SMS2FA. Unfortunately, I do not have a Mac (which is also the reason for the project ;) ), so I cannot test it here. Statements like "I have the same problem" do not really help here. I need at least the host OS and which shell is being used, to narrow down the error. 
An alternative would be to try running the endpoint natively, without Docker (python3 have to be installed).

```
git clone https://github.com/dchristl/macless-haystack.git
cd macless-haystack/endpoint/
pip install --no-cache-dir -r requirements.txt
python3 mh_endpoint.py
```

That is the same thing the container is doing. 

@a-camacho 

Normally it should work, but you can also follow my instructions and try again. Although I don't think the errors are related (Apple's account management is extremely opaque, determining which account works and which doesn't), it might help to narrow down the issue.

fachinformatiker, 2024-03-30:
> Statements like "I have the same problem" do not really help here. I need at least the host OS and which shell is being used, to narrow down the error. An alternative would be to try running the endpoint natively, without Docker (python3 have to be installed).

Sorry, I was on the go, so I couldn't provide more informations.
I'm using Debian 12 on a server.
Running your container on this server outputs the multiple lines of text.

the code for running it local gives me also an error. :(
> 
> ```
> ModuleNotFoundError: No module named 'Crypto'
> ```
>

dchristl, 2024-03-31:
> I'm using Debian 12 on a server with the default bash shell.
Running your container on this server outputs the multiple lines of text.

Thank you for your answer. I'm using several Debians or Ubuntu Server, because this is my preferred system. I will install  a fresh one and retry it. 
Are you connected to this server by ssh or directly (with a physical keyboard). If with ssh, what shell/client are you using for connecting? Ist this system virtualized (VMWare, VirtualBox)? Is this system up to date (latest updates, docker)? 

> ModuleNotFoundError: No module named 'Crypto'

This will be normally imnstalled by pip install --no-cache-dir -r requirements.txt. Was there an error by this command? Are there multiple python installations on your system? Alternatively you can try `python3 -m pip install --no-cache-dir -r requirements.txt`

coopeeo, 2024-03-31:
i have the same problem running on docker and on windows

fachinformatiker, 2024-04-01:
I've tried running this command:
> 
> ```
> docker run -it --restart unless-stopped --name macless-haystack -p 6176:6176 --volume mh_data:/app/endpoint/data --network mh-network christld/macless-haystack:latest-dev
> ```
> 

but still get errors :(
Tried it on my RasPi4 now, running bash. I'm connecting to it like to my server via the App Termius.

```
Domain=gsa.apple.com; Path=/; Secure; HttpOnly, site=USA; Domain=apple.com; Path=/; Secure; HttpOnly
Pragma: no-cache
Expires: Thu, 01 Jan 1970 00:00:00 GMT
Cache-Control: no-cache, no-store
X-Apple-I-Rscd: 412
vary: accept-encoding
Content-Encoding: gzip
Content-Language: en-US-x-lvariant-USA
Keep-Alive: timeout=30

Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 159, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 77, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 50, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 40, in icloud_login_mobileme
    g = gsa_authenticate(username, password)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 121, in gsa_authenticate
    sms_second_factor(spd["adsid"], spd["GsIdmsToken"])
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 276, in sms_second_factor
    raise Exception(
Exception: 2FA unsuccessful. Maybe wrong code or wrong number. Check your account details.
```

Weird thing is, that I don't get the 2FA Popup and need to go into settings and copy a code from there. :(
I don't really want to create a new Apple ID xD
All I want is to log the location of my tags (to see where they went -> for my bicycles)

JJTech0130, 2024-04-01:
> When lauching macless-haystack I see a strange behavior, maybe this is related :
> 
> > remote: Enumerating objects: 88, done.
> > remote: Counting objects: 100% (88/88), done.
> > remote: Compressing objects: 100% (47/47), done.
> > remote: Total 75 (delta 37), reused 56 (delta 25), pack-reused 0
> > Unpacking objects: 100% (75/75), 3.62 MiB | 18.26 MiB/s, done.
> > From [dchristl/macless-haystack](https://github.com/dchristl/macless-haystack)
> > branch            main       -> FETCH_HEAD
> > [32ab133](https://github.com/dchristl/macless-haystack/commit/32ab13394cb4ec553b35cc090574f6efc76abb16)..[e2ad25c](https://github.com/dchristl/macless-haystack/commit/e2ad25cd513fe074a65ae5ac94dc7395023c2900)  main       -> origin/main
> > 2024-03-04 17:46:54,267 - INFO - No auth-token found.
> > 2024-03-04 17:46:54,268 - INFO - Trying to register new device.
> > Apple ID: [apple@mpalandre.fr](mailto:apple@mpalandre.fr)
> > Password:
> > 2024-03-04 17:47:20,772 - INFO - 2FA required, requesting code
> > 2024-03-04 17:47:22,892 - INFO - 2FA required, requesting code
> > Enter 2FA code: 326094
> > 2024-03-04 17:48:02,547 - INFO - 2FA successful
> > 2024-03-04 17:48:04,620 - INFO - 2FA required, requesting code
> > Enter 2FA code:
> 
> Here is what is strange :
> 
> 1. As you can see there are 2 lines saying its requesting 2FA. Maybe this is why even if I enter the 2FA, it's still asking for the 2nd 2FA ?
> 2. Even if I enter a random 2FA I get at least one line saying `2FA successful`

Asking multiple times could also be caused by several things, probably Apple's server returning something odd in response to the SMS that we interpret as success, when success is received in response to 2FA submission we essentially just make the initial login request again, if 2FA was really successful then it doesn't prompt for a second factor the second time.

dchristl, 2024-04-02:
> Exception: 2FA unsuccessful. Maybe wrong code or wrong number. Check your account details.

Yes, that's an intentional exception, and that's the new error handling I've implemented. Apple doesn't accept your 2FA, so you can't log in. The error description now also states that you mistyped or your data with Apple is incorrect.

 > Weird thing is, that I don't get the 2FA Popup and need to go into settings and copy a code from there.

What kind of pop-up are you expecting? Where are you copying any 2FA code from? You must receive a text message, only codes from SMS are accepted.Did you get a message?

 

> Asking multiple times could also be caused by several things, probably Apple's server returning something odd in response to the SMS that we interpret as success, when success is received in [...]

That happens doubly whenever the 2FA-code is incorrect; on the next retrieval, it's requested again. There's an error in the original script from biemster, which always outputs "2FA successful". I've added an additional check in the dev branch to immediately abort (see above).

JJTech0130, 2024-04-02:
> What kind of pop-up are you expecting? Where are you copying any 2FA code from? You must receive a text message, only codes from SMS are accepted.Did you get a message?

They’re trying to use trusted device 2FA (using a logged in Mac or iPhone). Need the `--trusteddevice` flag

dchristl, 2024-04-02:
> They’re trying to use trusted device 2FA (using a logged in Mac or iPhone). Need the --trusteddevice flag with the original script, not sure if you expose that functionality.

This is not implemented in macless haystack anymore, because you need a real device (the goal of this project is to not need a Mac) and it's also written in the documentation (prerequisites) that a SMS 2FA is required. I think I will clear this in documentation and inside the endpoint, if this is the problem.

coopeeo, 2024-04-02:
I don't receive any SMS messages at all and I'm not sure what the problem is.

dchristl, 2024-04-02:
> I don't receive any SMS messages at all and I'm not sure what the problem is.

Can you at https://appleid.apple.com/ with SMS2FA? Otherwise check you number. Sometimes Apple blocks the sending if it is requested too often. This is visible in the response from Apple, if you use the latest-dev-Docker container.

coopeeo, 2024-04-02:
okay I will look at it later today

fachinformatiker, 2024-04-02:
How can I add "--trusteddevice"?
openhaystack os not working on my new macbook, that's why I want to host it on my server.
But I have Apple devices, that's why I want to use "--trusteddevice"
Thanks

JJTech0130, 2024-04-02:
https://github.com/dchristl/macless-haystack/blob/335675d9718cf72c6660149f4ff9ddf852e703db/endpoint/mh_endpoint.py#L98
This line is where he hardcodes it to `sms`, you'll have to edit it inside the Docker container, change it to `trusted_device`

fachinformatiker, 2024-04-02:
> This line is where he hardcodes it to `'sms'`, you'll have to edit it inside the Docker container, change it to `'trusted_device'`

I stopped the container, changed the line, rerun it and I'm still not getting the Token =(
I've tried the default image and the dev-image

dchristl, 2024-04-05:
> I stopped the container, changed the line, rerun it and I'm still not getting the Token =(
I've tried the default image and the dev-image

Why don't you simply add a mobile number to your account?

fachinformatiker, 2024-04-05:
> Why don't you simply add a mobile number to your account?

been there, done that.
I have a mobile number linked to my account, but I never get a SMS

aaronjamt, 2024-04-06:
I'm having the same issue. I tried the patch [@JJTech0130 suggested](https://github.com/dchristl/macless-haystack/issues/46#issuecomment-2033079950) but still didn't get a code. I was able to log in by cloning https://github.com/biemster/FindMy, pip installing `pbkdf2` and `srp`, then running `python3 request_reports.py -t`. When that completed, I copied the `auth.json` file into the macless-haystack container at `/app/endpoint/data/auth.json`, and restarting the container. I used the same anisette container, using the command in the `README.md`, so that eliminates that as the issue. However, I also did not receive an SMS using that method (before I added `-t` for authentication using a trusted device), so I don't think the SMS issue is the fault of this repo. I'd still suggest adding a flag to use trusted devices and trying to fix that, but the main point of this issue (the SMS error) seems to be upstream.

JJTech0130, 2024-04-08:
The one thing that I see is that we assume you always want to send the SMS to the first phone number registered to the account: you don't happen to have multiple phone numbers attached to your account, do you?

dchristl, 2024-04-12:
Is it possible for any of you to login at https://appleid.apple.com/ and manually select the sms as second factor? Does this text reach you?

a-camacho, 2024-04-14:
> Is it possible for any of you to login at https://appleid.apple.com/ and manually select the sms as second factor? Does this text reach you?

Personally, I can login on icloud.com and manually select SMS (saying "I didn't receive the code") and yes, I can get the SMS right. But not from `macless-haystack`

mg8x, 2024-04-19:
i also tried on my ubnutu vps server i dont get the 2FA code SMS.
note:
i have tried on my windows real machine it worked.
my apple id is linked to a phone number.
i reseit in my vps multiple times but the same.
i copied the auth file from my windows to ubnutu but still not working.

dchristl, 2024-04-21:
@a-camacho:
This is very strange. I'm sorry, I don't know how to proceed from here and can't do anything about it. I can't reproduce or explain the issue. Please try the workaround eplained at [this comment](https://github.com/dchristl/macless-haystack/issues/46#issuecomment-2030128287)

@mg8x :
could you provide a [log](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-see-the-logs), the copy works normally.

mg8x, 2024-04-21:
here is the log:
note : i entered my email and password then it asked for 2FA but the sms didnt come so i entered something random first time it asked again i enetered none then asked again i pressed Ctrl+C to exit

`remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 11 (delta 4), reused 8 (delta 4), pack-reused 0
Unpacking objects: 100% (11/11), 6.82 KiB | 873.00 KiB/s, done.
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
   67a74ff..335675d  main       -> origin/main
2024-04-21 19:06:15,625 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-04-21 19:06:15,626 - INFO - No auth-token found.
2024-04-21 19:06:15,626 - INFO - Trying to register new device.
Apple ID: *****@gmail.com
Password:
2024-04-21 19:06:37,468 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:06:37,471 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:06:37,484 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:06:37,487 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:06:38,288 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1356
2024-04-21 19:06:38,885 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:06:38,886 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:06:38,888 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:06:38,891 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:06:39,724 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 5232
2024-04-21 19:06:39,823 - INFO - 2FA required, requesting code
2024-04-21 19:06:39,831 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:06:39,833 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:06:39,838 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:06:39,840 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:06:40,785 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1356
2024-04-21 19:06:41,416 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:06:41,417 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:06:41,422 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:06:41,424 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:06:42,435 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 5232
2024-04-21 19:06:42,442 - INFO - 2FA required, requesting code
2024-04-21 19:06:42,442 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:06:42,443 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:06:42,447 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:06:42,451 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:06:43,467 - DEBUG - https://gsa.apple.com:443 "PUT /auth/verify/phone/ HTTP/1.1" 200 None
Enter 2FA code: 123344^H^H
2024-04-21 19:07:00,320 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:07:01,317 - DEBUG - https://gsa.apple.com:443 "POST /auth/verify/phone/securitycode HTTP/1.1" 200 None
2024-04-21 19:07:01,319 - INFO - 2FA successful
2024-04-21 19:07:01,328 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:07:01,329 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:07:01,332 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:07:01,334 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:07:02,077 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 1356
2024-04-21 19:07:02,789 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:07:02,791 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:07:02,794 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:07:02,796 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:07:03,678 - DEBUG - https://gsa.apple.com:443 "POST /grandslam/GsService2 HTTP/1.1" 200 5232
2024-04-21 19:07:03,681 - INFO - 2FA required, requesting code
2024-04-21 19:07:03,681 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:07:03,682 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:07:03,692 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
2024-04-21 19:07:03,694 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:07:04,681 - DEBUG - https://gsa.apple.com:443 "PUT /auth/verify/phone/ HTTP/1.1" 200 None
Enter 2FA code: none
2024-04-21 19:07:07,420 - DEBUG - Starting new HTTPS connection (1): gsa.apple.com:443
2024-04-21 19:07:08,369 - DEBUG - https://gsa.apple.com:443 "POST /auth/verify/phone/securitycode HTTP/1.1" 200 None
2024-04-21 19:07:08,370 - INFO - 2FA successful
2024-04-21 19:07:08,379 - DEBUG - Querying http://anisette:6969 for an anisette server
2024-04-21 19:07:08,380 - DEBUG - Starting new HTTP connection (1): anisette:6969
2024-04-21 19:07:08,382 - DEBUG - http://anisette:6969 "GET / HTTP/1.1" 200 566
[... 20 lines cut ...]
               ^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
    sms_second_factor(spd["adsid"], spd["GsIdmsToken"])
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 293, in sms_second_factor
    code = input("Enter 2FA code: ")
           ^^^^^^^^^^^^^^^^^^^^^^^^^
KeyboardInterrupt
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-04-21 19:07:15,979 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-04-21 19:07:15,979 - INFO - No auth-token found.
2024-04-21 19:07:15,979 - INFO - Trying to register new device.
`

dchristl, 2024-04-22:
Thanks for the log @mg8x, but it is not the [other branch, like suggested](https://github.com/dchristl/macless-haystack/issues/46#issuecomment-2028702273). Nevermind, this is in the latest version since today. It seems that some accounts simply not work and I don't know really why. The current workaround is to register it with bimester's project (described [here](https://github.com/dchristl/macless-haystack/issues/46#issuecomment-1951343857)) and copy the auth.json to this project.

mg8x, 2024-04-22:
I dont know but i think the problem is im from middle east and my account registered there and my linux vps is in Germany so this might be the problem. Thank you i will try the solution you have suggested.

mg8x, 2024-04-22:
just now tried the bimester's it was same i didnt get any sms. i will try later with another apple id. thank you for your hard work

dchristl, 2024-04-22:
> just now tried the bimester's it was same i didnt get any sms. i will try later with another apple id. thank you for your hard work

You should try with [trusteddevice](https://github.com/dchristl/macless-haystack/issues/46#issuecomment-2030128287), if you have a real Apple device.

mg8x, 2024-04-22:
i will try it later. it would be great if it works on my VPS bcz i'm planning to develop a telegram bot for the reports(its too easy. and i would like to contribute to this project for the bot if you wish) and i think its better for getting notified.

Ramenaru, 2024-04-25:
im becoming elektronomia sky high, my 2FA setting isn't work lmao.

poseiso, 2024-04-25:
I am currently facing the same issue, should i show log as well?

Ramenaru, 2024-04-30:
hey i cannot get to the docker server when 2FA getting responded

Ramenaru, 2024-04-30:
i after success 2FA with sms, but my apple id is blocked why ?

log:

2024-04-30 10:01:46,307 - INFO - 2FA successful
2024-04-30 10:01:46,333 - INFO - Authentication request initialization
2024-04-30 10:01:48,259 - DEBUG - HTTP-Code: 200
---
 [2024-04-30 10:01:51,966 - DEBUG - Answer from icloud login
2024-04-30 10:01:51,966 - DEBUG - {'dsid': '21623485671', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'A server problem is blocking Apple ID sign in.]

dschense, 2024-05-03:
@Ramenaru 
do you try to login with enabled VPN?
Apple Login sometimes hates any type of VPN. If you are using, disable, login and enable again after login.

Ramenaru, 2024-05-05:
> @Ramenaru do you try to login with enabled VPN? Apple Login sometimes hates any type of VPN. If you are using, disable, login and enable again after login.

no I didn't use any VPN, or kinda like that. that's just say try adding a credit card, but my apple account is already have some credit card. Im from southeast Asia is that any related problem to my region ? or any suggest ?
please help

thirstyone, 2024-06-12:
yet another person who doesn't receieve an sms from apple (phone number registered (android phone, if that mattters), 2FA enabled, no vpn is used.. tried ... christld/macless-haystack:latest-dev - no luck ;-(

DevZombi, 2024-06-13:
Can report the same here. Not receiving the SMS text. Tried 2 accounts, both mature good standing accounts. Neither received the SMS.

maplepy, 2024-06-17:
No 2FA code is received here either

dchristl, 2024-06-17:
If all the tips in the FAQ don't help, then only Apple knows why the accounts aren't working. Sorry, there's nothing I can do here, so I'm closing the issue.

zerog2k, 2024-09-16:
> The other thing you could possibly do is, if you have a real mac logged into the account, use the anisette headers from the mac to bypass 2FA.

Can someone explain how to do this if we are able to login with 2FA to https://appleid.apple.com/account/manage, but not with macless-openhaystack ?

Xopher00, 2024-09-26:
I am having the exact same issue and am very frustrated this has not been resolved.
Excerpt from docker logs:
`remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 6 (delta 0), reused 5 (delta 0), pack-reused 0 (from 0)
Unpacking objects: 100% (6/6), 4.43 KiB | 1.11 MiB/s, done.
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
   92d56a0..7fcbea6  main       -> origin/main
2024-09-26 16:53:46,576 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-09-26 16:53:46,577 - INFO - No auth-token found.
2024-09-26 16:53:46,577 - INFO - Trying to register new device.
2024-09-26 16:53:46,578 - INFO - Trying to login
Apple ID: pricekr
Password:
2024-09-26 16:53:55,764 - INFO - Authentication request initialization
2024-09-26 16:53:56,545 - DEBUG - HTTP-Code: 200
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Response</key>
    <dict>
        <key>Status</key>
        <dict>
            <key>hsc</key>
            <integer>200</integer>
            <key>ed</key>
            <string></string>
            <key>ec</key>
            <integer>0</integer>
            <key>em</key>
            <string></string>
            <key>tt</key>
            <string></string>
            <key>rsh</key>
            <false/>
        </dict>
        <key>i</key>
        <integer>818</integer>
        <key>s</key>
        <data>5DqZSkLZYQtbiWIDgyV6IQ==</data>
        <key>sp</key>
        <string>s2k_fo</string>
        <key>ptxid</key>
        <string>77315882-3856-4714-ab69-139863107f46</string>
        <key>c</key>
        <string>d-53e-eb437572-7c27-11ef-8f3e-b7525ea82019:RNO</string>
        <key>B</key>
        <data>T1HkpuKbqMnXSy/6dCWtqPLehZxT9LTU/BymhfpIqQIJ8JRYysZEa/jumw7EdXgYHkOHef/Kt/inTDPb/1HpPQa6ufVLHTK8KKkImEjZYmKBRp4qaFxVNWPEnFhXxhLkxX97PCwWVcZe9Irx61w2yTI5IsP6GWgr5t7GeSE+f+8akmTpz75osDXhciFHyRH7emoJfXKPDmDRs4qimHx4K1nNFETJGwcCgknTnt1FWl07aIMV5VAD5Bq8tDd50hBRpknRGFwwcOYxsKcEICIk2FvZw+Z0R3+pLkQecwKCBs+aG1XTq36KQP3QmWeMhiol2WWk956B9w/dOj6Z0XCnqA==</data>
    </dict>
    <key>Header</key>
    <dict>
    </dict>
</dict>
</plist>

2024-09-26 16:53:56,548 - WARNING - This implementation only supports s2k. Server returned s2k_fo
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 159, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 77, in registerDevice
    getAuth(regenerate=True)
  File "/app/endpoint/register/apple_cryptography.py", line 50, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 41, in icloud_login_mobileme
    pet = g["t"]["com.apple.gs.idms.pet"]["token"]
          ~^^^^^
TypeError: 'NoneType' object is not subscriptable
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-09-26 16:54:00,466 - DEBUG - Searching for token at /app/endpoint/data/auth.json
2024-09-26 16:54:00,467 - INFO - No auth-token found.
2024-09-26 16:54:00,467 - INFO - Trying to register new device.
2024-09-26 16:54:00,467 - INFO - Trying to login`

dmickeyus, 2025-01-11:
> I am having the exact same issue and am very frustrated this has not been resolved. I also have sms set up for 2fa. This still has not been resolved? I was doing my own research and everything I found pointed to here as the easiest way to set things up. Excerpt from docker logs:
> 
> ```
> `remote: Enumerating objects: 6, done.
> remote: Counting objects: 100% (6/6), done.
> remote: Compressing objects: 100% (6/6), done.
> remote: Total 6 (delta 0), reused 5 (delta 0), pack-reused 0 (from 0)
> Unpacking objects: 100% (6/6), 4.43 KiB | 1.11 MiB/s, done.
> From https://github.com/dchristl/macless-haystack
>  * branch            main       -> FETCH_HEAD
>    92d56a0..7fcbea6  main       -> origin/main
> 2024-09-26 16:53:46,576 - DEBUG - Searching for token at /app/endpoint/data/auth.json
> 2024-09-26 16:53:46,577 - INFO - No auth-token found.
> 2024-09-26 16:53:46,577 - INFO - Trying to register new device.
> 2024-09-26 16:53:46,578 - INFO - Trying to login
> Apple ID: pricekr
> Password:
> 2024-09-26 16:53:55,764 - INFO - Authentication request initialization
> 2024-09-26 16:53:56,545 - DEBUG - HTTP-Code: 200
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <dict>
>     <key>Response</key>
>     <dict>
>         <key>Status</key>
>         <dict>
>             <key>hsc</key>
>             <integer>200</integer>
>             <key>ed</key>
>             <string></string>
>             <key>ec</key>
>             <integer>0</integer>
>             <key>em</key>
>             <string></string>
>             <key>tt</key>
>             <string></string>
>             <key>rsh</key>
>             <false/>
>         </dict>
>         <key>i</key>
>         <integer>818</integer>
>         <key>s</key>
>         <data>5DqZSkLZYQtbiWIDgyV6IQ==</data>
>         <key>sp</key>
>         <string>s2k_fo</string>
>         <key>ptxid</key>
>         <string>77315882-3856-4714-ab69-139863107f46</string>
>         <key>c</key>
>         <string>d-53e-eb437572-7c27-11ef-8f3e-b7525ea82019:RNO</string>
>         <key>B</key>
>         <data>T1HkpuKbqMnXSy/6dCWtqPLehZxT9LTU/BymhfpIqQIJ8JRYysZEa/jumw7EdXgYHkOHef/Kt/inTDPb/1HpPQa6ufVLHTK8KKkImEjZYmKBRp4qaFxVNWPEnFhXxhLkxX97PCwWVcZe9Irx61w2yTI5IsP6GWgr5t7GeSE+f+8akmTpz75osDXhciFHyRH7emoJfXKPDmDRs4qimHx4K1nNFETJGwcCgknTnt1FWl07aIMV5VAD5Bq8tDd50hBRpknRGFwwcOYxsKcEICIk2FvZw+Z0R3+pLkQecwKCBs+aG1XTq36KQP3QmWeMhiol2WWk956B9w/dOj6Z0XCnqA==</data>
>     </dict>
>     <key>Header</key>
>     <dict>
>     </dict>
> </dict>
> </plist>
> 
> 2024-09-26 16:53:56,548 - WARNING - This implementation only supports s2k. Server returned s2k_fo
> Traceback (most recent call last):
>   File "/app/endpoint/mh_endpoint.py", line 159, in <module>
>     apple_cryptography.registerDevice()
>   File "/app/endpoint/register/apple_cryptography.py", line 77, in registerDevice
>     getAuth(regenerate=True)
>   File "/app/endpoint/register/apple_cryptography.py", line 50, in getAuth
>     mobileme = icloud_login_mobileme(
>                ^^^^^^^^^^^^^^^^^^^^^^
>   File "/app/endpoint/register/pypush_gsa_icloud.py", line 41, in icloud_login_mobileme
>     pet = g["t"]["com.apple.gs.idms.pet"]["token"]
>           ~^^^^^
> TypeError: 'NoneType' object is not subscriptable
> From https://github.com/dchristl/macless-haystack
>  * branch            main       -> FETCH_HEAD
> 2024-09-26 16:54:00,466 - DEBUG - Searching for token at /app/endpoint/data/auth.json
> 2024-09-26 16:54:00,467 - INFO - No auth-token found.
> 2024-09-26 16:54:00,467 - INFO - Trying to register new device.
> 2024-09-26 16:54:00,467 - INFO - Trying to login`
> ```

I am a new joiner and found this project is interesting me.  I faced the exactly same issue when creating this docker app.  I believe Apple may changed the login authen method and thats why is not working.  The following diff on pypush_gsa_icloud.py let me authen sucessfully and functioning properly.  I wish this could be merged into the code.

---

79c80
<     if r["sp"] != "s2k":
---
>     if r["sp"] not in ["s2k", "s2k_fo"]:
85c86
<     usr.p = encrypt_password(password, r["s"], r["i"])
---
>     usr.p = encrypt_password(password, r["s"], r["i"], r["sp"])
193c194
< def encrypt_password(password, salt, iterations):
---
> def encrypt_password(password, salt, iterations, protocol):
194a196,197
>     if protocol == "s2k_fo":
>         p = p.hex().encode("utf-8")
278,279d280
<
<

[7 more replies]