dchristl/macless-haystack #53: Specify to which IP address the server binds

bartoszbielawski, 2024-03-12:
I would like to use a reverse proxy I already have on my server to handle HTTPS traffic.
That's why I'd prefer to be able to bind only to localhost. A config option would be handy for this.

dchristl, 2024-03-18:
Hello @bartoszbielawski ,

I will add this option to the next release.

bartoszbielawski, 2024-03-19:
Hi,

I'm an embedded programmer and not a web-guy, so please forgive my lack of knowledge.

Your solution does bind the HTTP server to the address that's specified, but it only binds it within the container, doesn't it?
This means that the following command (not sure how Docker generates it):
` /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 6176 -container-ip 172.18.0.3 -container-port 6176`

Isn't really what I wanted because:
1. it still binds to the local address of the container instead of the localhost address of the container (does a container have a localhost address?),
2. The ip:port from the container is still bound to all the addresses of the host.

Shouldn't it be just the docker-proxy that binds to the localhost instead of 0.0.0.0?

dchristl, 2024-03-20:
Hi, I'm not really a network guy. ;) Yes, that binds only within the container to the interfaces. I didn't consider that, and it only works natively without Docker. Unfortunately, I can't help you much there because my networking knowledge is limited. I think you would need to configure something in Docker to achieve what you want. I don't know what I would need to change in the application/container for that to work.

bartoszbielawski, 2024-03-20:
After reading I think I have found the answer.

When the container is created you specify port mappings using `-p HP:CP` format or using `-p H:HP:CP` format. The second one accepts IP address on the host side. Otherwise it binds to all interfaces.

I think that:
1. The anisette (whatever it does) should probably NOT be exposed beyond the container. My guess is that in that case `-p` is not only not needed, it probably should not be there. 
2. For the macless-haystack container one can create local-only binding at that moment. Not possible to modify it later!

I'd suggest checking this and improving README file.

dchristl, 2024-03-20:
> When the container is created you specify port mappings using -p HP:CP format or using -p H:HP:CP format. The second one accepts IP address on the host side. Otherwise it binds to all interfaces.

Yes, that makes sense. So, simply specify "localhost:6176", and then the server will not be externally accessible, and everything will work with the reverse proxy.

> The anisette (whatever it does) should probably NOT be exposed beyond the container. My guess is that in that case -p is not only not needed, it probably should not be there.

The Anisette server provides (generated) access credentials required by the macless haystack server. So, the Haystack server needs to access it. While this could work differently (using an internal Docker network), it would complicate the configuration.

> For the macless-haystack container one can create local-only binding at that moment. Not possible to modify it later!

That's the way Docker operates. You can either modify the configuration or simply delete the container `docker rm -f macless-haystack` and recreate it with the other command. All data will be preserved in the process.