dchristl/macless-haystack #56: It seems your account score is not high enough.

GameEnder, 2024-03-14:
Any tips for getting it to connect correctly with my AppleID? I have has this ApplelD since 2012 and have added a credit card already and it it is still not working after several attempts. I fully removed both containers after each attempt.

Any other things I can do the get my account is good enough standing to get this to work?

GameEnder, 2024-03-14:
Resolved. Apple dose not seem to like if you don't buy any new devices for a while. Added a iPad from my pile of extras and now my account works.

closed 2024-03-14 by GameEnder

dchristl, 2024-03-18:
That's the unpredictability of Apple's policies. How and when accounts are approved or denied is something you can only figure out through trial and error.