dchristl/macless-haystack #58: BBC MicroBit V2 Flashing

GameEnder, 2024-03-16:
How do you go about flashing the BBC MicroBit V2 with the NRF52? I am not finding much documentation on installing firmware on the device. Just hex compiled programs.

GameEnder, 2024-03-16:
Here is the answer to my own question. This is how to setup and flash it on windows, but pyocd should also work on linux.

#Install Scoop
Set-ExecutionPolicy -ExecutionPolicy RemoteSigned -Scope CurrentUser
Invoke-RestMethod -Uri https://get.scoop.sh | Invoke-Expression

#Install pipx (Reload Powershell Afterwards)
scoop install pipx
pipx ensurepath

#Install pyocd with pipx 
pipx install -U pyocd

#Install Firmware
pyocd erase --mass --t nrf52833
pyocd flash -t nrf52833 C:\%Folder%\nrf52_firmware.bin

dchristl, 2024-03-18:
Thank you for figuring it out yourself. As I've written in the documentation, the NRF52 is untested because I don't have such a chip, but theoretically, it should work.