dchristl/macless-haystack #61: Cannot access server on the local network (missing "Access-Control-Allow-Private-Network: true" header).

olivluca, 2024-03-20:
If I'm using your frontend hosted on github, I cannot access the server on the local network (or on localhost) because:

`Access to XMLHttpRequest at 'https://localhost:6176/' from origin 'https://dchristl.github.io' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Private-Network' header was present in the preflight response for this private network request targeting the `local` address space.
`
This is on chromium, firefox reports the same issue but less verbosely:

`Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://localhost:6176/. (Reason: CORS request did not succeed). Status code: (null).
`

dchristl, 2024-03-22:
Hello,

the problem is that you are using https on your local machine instead of http. This is normally not needed and causes this error. You connect to your own machine with SSL encryption, which is useless. The browsers block this.
SSL should onyl be used if you have your own server.

Regards,
Danny

olivluca, 2024-03-22:
Well, ssl was a last ditch effort to see if it solved the problem. It's the same with plain http. It's a safety feature, to avoid that an external site (in this case github) can access internal resources, unless the internal resource allows access with the "Access-Control-Allow-Private-Network:true". At least that's what I understood reading the documentation (which I didn't study thoroughly so it's very well possible that I didn't understand correctly how to solve the problem).
I guess that if I host the page internally I would solve the problem.

olivluca, 2024-03-22:
(btw: localhost was the second attempt with an ssh tunnel since I got the error accessing the real, internal, machine, which is not the same where I'm running the browser).

dchristl, 2024-03-23:
I know the CORS problem, but the header will already sent by the server:
[image]

This allows any host, including localhost. Can you check if this header is in the response? localhost is kind of special and will be handled different by browsers. The problem I mentioned with SSL is that some browsers have problems with self signed certificates on localhost. So, please avoid this with localhost.

 > (btw: localhost was the second attempt with an ssh tunnel since I got the error accessing the real, internal, machine, which is not the same where I'm running the browser).

Can you explain your setup? I don't understatnd the need of the ssh tunnel (port forwarding?). Your browser connects directly to the endpoint machine. So if your endpoint is running on your client machine, you have to use localhost othwise another IP (i.e 192.168.XX.XX for internal network)

olivluca, 2024-03-23:
The "Access-Control-Allow-Origin: *" is ther, what's missing is the "Access-Control-Allow-Private-Network: true", see here:
https://developer.chrome.com/blog/private-network-access-preflight

The setup is simple: I was pointing the frontend to another host in the internal network, got the error, tried localhost (with an ssh tunnel), same error, tried ssl.

dchristl, 2024-03-23:
> The "Access-Control-Allow-Origin: *" is ther, what's missing is the "Access-Control-Allow-Private-Network: true", see here:
https://developer.chrome.com/blog/private-network-access-preflight

I can set the header, no problem, but it's not necessary. I just tried it out. If your computer, where you are accessing the GitHub page, has the IP 192.168.1.10 and your endpoint has the IP 192.168.1.20, then you must use HTTPS (this is a browser restriction). HTTP is only allowed for localhost. You're probably using a self-signed certificate. In that case, you need to first access your endpoint in the browser (https://192.168.1.20:6176/) and accept the certificate. Then you can enter the URL in your client machine as the endpoint address, and everything should work.

This is also explained in the [FAQ](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-use-ssl-if-the-endpoint-runs-on-another-machine-than-the-ui)

olivluca, 2024-03-23:
Never mind. I added the header myself. Now the problem is that apple has disabled my id and I cannot create a new one right now.

olivluca, 2024-03-23:
Solved the issue with the apple id, I can now confirm that without the header it doesn't work, with the header it does.
I just added 

`self.send_header("Access-Control-Allow-Private-Network","true")`

to the end of the `addCORSHeaders` function.

olivluca, 2024-04-03:
I don't understand either, maybe a different version of the browser or something in about:config that makes it more lenient in your case. I also don't get 100% the security implications of adding the header but I don't think it would be a problem for this application.