dchristl/macless-haystack #63: Add Samsung SmartTag support

T-REX-XP, 2024-03-21:
Can we also add the support of the Samsung SmartTag . Here is the reference  to the implementation. https://github.com/MatthewKuKanich/FindMyFlipper 

It seems like the pairing mode will be supported compare to the air tag implementation.

https://www.reddit.com/r/flipperzero/comments/1b6xfsl/findmy_flipper_location_tracking_using_the_findmy/

dchristl, 2024-03-22:
As mentioned in the Readme, all devices compatible with FindMy or openHaystack also work with Macless Haystack. No special firmware is required for that.

Your device is compatible to FindMy, so it should work already.

ngothiensinh, 2024-04-04:
I am still quite confused about run the project locally, especially about flashing firmware ... Could you please make a video for how to run it?

I have bought some Tags that produced by 3-rd company. And I would like to create custom web solution to monitor these Tags (My Dogs will wear these tag).
Does your solution will fit in my case, because I think we need to get the Private key to descrypt the position, And I am not sure it is possible.

@dchristl, please help

dchristl, 2024-04-05:
> I have bought some Tags that produced by 3-rd company. And I would like to create custom web solution to monitor these Tags (My Dogs will wear these tag).

Do projects like FindMy or OpenHaystack support these tags? If so, you should be able to find a suitable firmware for these projects.

>Does your solution will fit in my case, because I think we need to get the Private key to descrypt the position, And I am not sure it is possible.

Probably not 100%. If you want to build your own frontend, then Macless Haystack is probably not suitable. However, you can take a look [here](https://github.com/biemster/FindMy) (this is the basis for this project).