dchristl/macless-haystack #68: Forward positions to 3rd-Party-Server

dschense, 2024-03-27:
Amazing Projekt. Working like a charme.
Perhaps integrating forwarding to another server would be a nice idea. For example, Traccar: https://github.com/traccar/traccar would be a very nice variant for this. This would allow you to manage all tracking devices under one platform.

Systm21, 2024-03-27:
https://github.com/mrmay-dev/owntags

dchristl, 2024-03-28:
Hello @dschense ,

I'm sorry, I don't really see that feature in this project. Architecturally, it wouldn't work well either because for security reasons, the server and the keys are distributed separately. Forwarding would then have to be done from the frontend or the app, as only there is the possibility of decryption. However, feel free to customize and use the sources according to your wishes. But probably the project linked by systm21 or the one by [biemster](https://github.com/biemster/FindMy) would be better suited as a basis.

Systm21, 2024-03-29:
> @dschense just write a proxy that grabs coords from this and puts it into traccar.

If it were that easy, or if he had programming skills, he would surely have done it already

closed 2024-04-22 by dchristl