dchristl/macless-haystack #70: Macless_Haystack stopped working

isibizi, 2024-04-12:
I cant login anymore in the macless haystack container. I have following error message

`remote: Enumerating objects: 11, done.
remote: Counting objects:   9% (1/11)
remote: Counting objects:  18% (2/11)
remote: Counting objects:  27% (3/11)
remote: Counting objects:  36% (4/11)
remote: Counting objects:  45% (5/11)
remote: Counting objects:  54% (6/11)
remote: Counting objects:  63% (7/11)
remote: Counting objects:  72% (8/11)
remote: Counting objects:  81% (9/11)
remote: Counting objects:  90% (10/11)
remote: Counting objects: 100% (11/11)
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects:  14% (1/7)
remote: Compressing objects:  28% (2/7)
remote: Compressing objects:  42% (3/7)
remote: Compressing objects:  57% (4/7)
remote: Compressing objects:  71% (5/7)
remote: Compressing objects:  85% (6/7)
remote: Compressing objects: 100% (7/7)
remote: Compressing objects: 100% (7/7), done.
remote: Total 11 (delta 4), reused 8 (delta 4), pack-reused 0
Unpacking objects:   9% (1/11)
Unpacking objects:  18% (2/11)
Unpacking objects:  27% (3/11)
Unpacking objects:  36% (4/11)
Unpacking objects:  45% (5/11)
Unpacking objects:  54% (6/11)
Unpacking objects:  63% (7/11)
Unpacking objects:  72% (8/11)
Unpacking objects:  81% (9/11)
Unpacking objects:  90% (10/11)
Unpacking objects: 100% (11/11)
Unpacking objects: 100% (11/11), 6.82 KiB | 3.41 MiB/s, done.
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
   67a74ff..335675d  main       -> origin/main
2024-04-12 10:09:10,738 - INFO - No auth-token found.
2024-04-12 10:09:10,738 - INFO - Trying to register new device.
Apple ID: xxxxxxxxxxxxxxxxxxxxx
Password: 
Traceback (most recent call last):
  File "/app/endpoint/mh_endpoint.py", line 158, in <module>
    apple_cryptography.registerDevice()
  File "/app/endpoint/register/apple_cryptography.py", line 76, in registerDevice
    getAuth(regenerate=True, second_factor='trusted_device' 'sms')
  File "/app/endpoint/register/apple_cryptography.py", line 49, in getAuth
    mobileme = icloud_login_mobileme(
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 40, in icloud_login_mobileme
    g = gsa_authenticate(username, password, second_factor)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 73, in gsa_authenticate
    r = gsa_authenticated_request(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 129, in gsa_authenticated_request
    "Request": {"cpd": generate_cpd()},
                       ^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 161, in generate_cpd
    cpd.update(generate_anisette_headers())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/register/pypush_gsa_icloud.py", line 169, in generate_anisette_headers
    f'Querying {config.getAnisetteServer()} for an anisette server')
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/endpoint/config.py", line 22, in getAnisetteServer
    value = config.get('Settings', 'anisette_url')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/configparser.py", line 759, in get
    d = self._unify_values(section, vars)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/configparser.py", line 1130, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'Settings'
From https://github.com/dchristl/macless-haystack
 * branch            main       -> FETCH_HEAD
2024-04-12 10:09:31,410 - INFO - No auth-token found.
2024-04-12 10:09:31,410 - INFO - Trying to register new device.`

dchristl, 2024-04-12:
> raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'Settings'

It seems your config.inifile is broken, invalid or not there anymore. This file is normally under `/var/lib/docker/volumes/mh_data/_data` You can either fix it there or simply delete it. After a container restart it should be recreated,

isibizi, 2024-04-12:
I have tried this allready, not working anymore. I have created a new docker env eirh the same issue.

dchristl, 2024-04-13:
Have you checked the file I mentioned? Is it there and valid? I think you haven't reset correct, if the issue still persist. You have to delete the volume,too. Otherwise your (propably corrupt) data stays intact. I've described it [here](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-reset-everything-and-start-over-how-can-i-completely-uninstall-macless-haystack)

isibizi, 2024-04-13:
@dchristl Thanks, it seems that the volumes were corrupt. I have deleted everything you described in your README, and now everything is working fine.

dchristl, 2024-04-13:
Great that it will work now.

closed 2024-04-13 by dchristl