dchristl/macless-haystack #71: Just A Question

wes1993, 2024-04-14:
Hello,
Is possible to track the original Apple AirTag using this?

I prefer to not install an entire OS just to track the devices.

Best Regards
Stefano

T-REX-XP, 2024-04-14:
It make sense to use third party smart tag taht will be supported by Google Find My Network. It's already released. The price is the same actually. https://www.androidpolice.com/search/?q=Google+find+my

dchristl, 2024-04-15:
Hello @wes1993 ,

this is not possible because you need the private key of the device. There are several tries to extract this from the Apple keychain, but not with real success currently. 

Kind Regards,
Danny

darthnithin, 2024-04-20:
Should be possible.
See https://github.com/malmeloo/FindMy.py
https://github.com/malmeloo/FindMy.py/blob/main/examples/real_airtag.py
and
https://github.com/malmeloo/FindMy.py/issues/4

dchristl, 2024-04-21:
Thank you for your post @darthnithin . My last update was that the extraction process is still ongoing, but it's taking some time due to the changing keys. Unfortunately, I don't have an AirTag, so I can't try it out myself.

wes1993, 2024-04-22:
Hello @dchristl and @darthnithin,
Thanks for replies, I have tried the malmelo and for now seems to work, not very well but it could be considered a POC.

Best Regards
Stefano

hkfuertes, 2024-07-13:
Hello, I'm trying to adapt (an use) Real AirTags too... and I have a question regarding all the keying (really really newbie)... I've got my plist and I can assure that it reports the correct location using FindMy.py... Now I was trying to """inject""" the plist into the android app, but I really don't know what goes where here:
```dart
  Accessory.fromJson(Map<String, dynamic> json)
      : id = json['id'],
        name = json['name'],
        hashedPublicKey = json['hashedPublicKey'],
        datePublished = json['datePublished'] != null
            ? DateTime.fromMillisecondsSinceEpoch(json['datePublished'])
            : null,
        _lastLocation = json['latitude'] != null && json['longitude'] != null
            ? LatLng(json['latitude'].toDouble(), json['longitude'].toDouble())
            : null,
        isActive = json['isActive'],
        isDeployed = json['isDeployed'],
        _icon = json['icon'],
        color = Color(int.parse(json['color'], radix: 16)),
        usesDerivation = json['usesDerivation'] ?? false,
        symmetricKey = json['symmetricKey'],
        lastDerivationTimestamp = json['lastDerivationTimestamp'],
        updateInterval = json['updateInterval'],
        oldestRelevantSymmetricKey = json['oldestRelevantSymmetricKey'],
        additionalKeys =
            json['additionalKeys']?.cast<String>() ?? List.empty() {
    _init();
  }
```
Here is the PLIST (redacted):
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>batteryLevel</key>
	<integer>1</integer>
	<key>cloudKitMetadata</key>
	<data> ... </data>
	<key>identifier</key>
	<string>...</string>
	<key>isZeus</key>
	<false/>
	<key>model</key>
	<string></string>
	<key>pairingDate</key>
	<date>2024-07-01T15:33:39Z</date>
	<key>privateKey</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			(28+ bit long hash)
			</data>
		</dict>
	</dict>
	<key>productId</key>
	<integer>21760</integer>
	<key>publicKey</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			...
			</data>
		</dict>
	</dict>
	<key>secondarySharedSecret</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
                          ...
			</data>
		</dict>
	</dict>
	<key>sharedSecret</key>
	<dict>
		<key>key</key>
		<dict>
			<key>data</key>
			<data>
			...
			</data>
		</dict>
	</dict>
	<key>stableIdentifier</key>
	<array>
		<string>...</string>
	</array>
	<key>systemVersion</key>
	<string>1.0.291</string>
	<key>vendorId</key>
	<integer>76</integer>
</dict>
</plist>
```

Is it possible to transform the plist into an acceptable json that could be imported? Or is the privatekey generated by the app? I know flutter and dart, and I can modify the app myself (I'm not seeking someone to do it for me), I just seek knowledge.

FindMy imports the PLIST here: https://github.com/malmeloo/FindMy.py/blob/90eaa6835c9ea49483be7a6d84e0faa1861427d1/findmy/accessory.py#L148

Thank you in advance!

adguy360, 2024-11-24:
So there's a problem extracting keys from original Airtags, what about all those cheap AliExpress ones that support the find my network? I'd assume they also use haystack for setting up the tags