dchristl/macless-haystack #72: can only obtain the location once a day

11wanqin, 2024-04-14:
I'm using the firmware of ESP32, with the service deployed on WSL, and everything is running smoothly. However, I can only obtain the location information approximately once a day. Below are the outputs from ESP32 and the service:
[image]
[image]

dchristl, 2024-04-15:
Hello @11wanqin ,

it seems there are some [changes ](https://github.com/espressif/esp-idf/issues/11804) in the setting of the mac-address (the firmware sets a random address on each key change). I will dig into it, when I have time. 

How many keys are you using?

11wanqin, 2024-04-16:
> Hello @11wanqin ,
> 
> it seems there are some [changes ](https://github.com/espressif/esp-idf/issues/11804) in the setting of the mac-address (the firmware sets a random address on each key change). I will dig into it, when I have time.
> 
> How many keys are you using?

Thank you for your reply. I used the key generated by generate_keys.py, so there should only be one.
Here is the data for the key:
[image]

dchristl, 2024-04-21:
It seems like your key or flashing process didn't work properly. Your generated key consists only of FF FF, which indicates a faulty structure. Consequently, the Mac address doesn't match either, as it's generated from the key. Honestly, I'm surprised it even works at all. Could you try flashing again from scratch and post the command and output?

supaeasy, 2024-05-08:
I created my first tag with more than one (10) key about 30 hours ago. Have had one location report since and that's it. I think there might be something wrong with either the firmware or key rotation recognition. Nrf51.

dschense, 2024-05-08:
for me there are no problems. have you tried to go to a place with some people? maybe there are not enough devices around you?

supaeasy, 2024-05-08:
I was at the office today carrying two trackers with me. One was updated very reliably and the other one not at all. Maybe a faulty device or corrosion, it got a little wet yesterday. Will do further testing over the weekend.

11wanqin, 2024-05-22:
> It seems like your key or flashing process didn't work properly. Your generated key consists only of FF FF, which indicates a faulty structure. Consequently, the Mac address doesn't match either, as [...]

Sorry, I've been a bit busy recently. After re-downloading the program to the ESP32 according to your suggestion, everything is running smoothly. The location information updates approximately every 20 minutes.
[image]
[image]

closed 2024-05-22 by 11wanqin