dchristl/macless-haystack #74: Anisette Updated Errors

GameEnder, 2024-04-19:
Looks like Anisette has a update and is now giving a not expected response for location request. 

```
Unknown error occured Expecting value: line 1 column 1 (char 0)

Traceback (most recent call last):

  File "/app/endpoint/mh_endpoint.py", line 103, in do_POST

    result = json.loads(r.content.decode())

             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.12/json/__init__.py", line 346, in loads

    return _default_decoder.decode(s)

           ^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.12/json/decoder.py", line 337, in decode

    obj, end = self.raw_decode(s, idx=_w(s, 0).end())

               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/local/lib/python3.12/json/decoder.py", line 355, in raw_decode

    raise JSONDecodeError("Expecting value", s, err.value) from None

json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

172.20.0.1 - - [19/Apr/2024 20:33:14] "POST /haystack HTTP/1.1" 501 -

```

dchristl, 2024-04-21:
Hello @GameEnder ,

if you have updated teh Anisette server on a formerly running system you have to reregister, described [here](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#restart-the-registrationchange-account)

GameEnder, 2024-04-21:
I have tried and it get stuck at "Trying to register new device."

dchristl, 2024-04-21:
OK, that would have been my next suggestion. It's great that it's working now.

closed 2024-04-21 by dchristl