dchristl/macless-haystack #76: some tag report not decrypted anymore

pheex, 2024-04-21:
Hello,

I'm getting an error for one of my tag that previously working : "Incorrect length for compressed encoding"
No error on backend side, so this is mostly in android app

Looks like apple changed something in reports : https://github.com/biemster/FindMy/issues/52

dchristl, 2024-04-21:
Hello @pheex ,

thanks for pointing this out. Is it possible to import your not working tag at the [web version](https://dchristl.github.io/macless-haystack/) ? I have imported the fix for this there. I fit will work I can release a new version. Thank you

GameEnder, 2024-04-21:
I have tried that and it get stuck at "Trying to register new device." Never go's any farther.

pheex, 2024-04-21:
locations are well decrypted 👍

closed 2024-04-22 by dchristl

pheex, 2024-04-22:
Sorry to reopen this topic, I don't know if this was the case yesterday when I tested the fix on the web version (I was focusing on the error msg which was no longer displayed with the new version), but now on the android or web version, even if the reports are decrypted without error, the positions are not displayed on the map.

dchristl, 2024-04-23:
Is it possible to post a screenshot? You can't see the last one or inside the history?