dchristl/macless-haystack #77: The cheapest and easiest device to flash that has been proofed to work?

mg8x, 2024-04-21:
Hi there i have read many issues about this topic and i have found some of them old that the product on aliexpress is not listed anymore or the flashing part so confusing. so i want to hear from anyone that did it in current time and worked without problem. i will be grateful if you provided the link to the product and share how you did it.

dschense, 2024-04-22:
This one is working: NRF51822
https://de.aliexpress.com/item/32826502025.html?gatewayAdapt=glo2deu

just a little search for this device and you can buy 2-3 for this single pricing.

dchristl, 2024-04-22:
Hello @mg8x ,

each device has its advantages and disadvantages. The NRF51822 chips are by far the most expensive and largest, and the batteries are relatively expensive as well. However, they have a long battery life. The ESP32 is affordable and versatile, but it's not available as a portable tracker (it needs to be soldered) and is very power-hungry. I use it for devices with their own power supply, such as my car. In my opinion, the best trackers are those with the ST17H66 chip. They are very affordable and usually come with batteries included. However, they are a bit more [challenging to flash](https://github.com/biemster/FindMy/issues/14), and sometimes you may get other chips like the ST17H66T, which currently cannot be flashed.

I have bought [these](https://de.aliexpress.com/item/1005005834064700.html) for example.

mg8x, 2024-04-22:
I have used the esp32 its very easy to flash but as you said its power hungry and its large to its purpose. 

For the ST17H66T as you said its challenging to flash I've read it and i was about to get headache from it and the first post link was not listed on aliexpress any more so it is like a gamble to buy random one especially when delivery for me costs more than the product it self it will worth if i bought 10 pcs at once. so i was wondering if anyone has already experienced with it so i can copy him.

> I have bought [these](https://de.aliexpress.com/item/1005005834064700.html) for example.

did it work?

mg8x, 2024-04-22:
Im so sorry to bother you and take your time but i would be grateful if you could share the devices, links and the process in easy detailed way especially the firmware part. Again sorry to bother you. It will be usefull for the community also!!

dchristl, 2024-04-23:
It is all in the link, I posted before. I also posted a link where you can buy the same I bought. I currently don't have time to write a better description.

supaeasy, 2024-04-24:
@mg8x I wrote a very detailed [step-by-step](https://github.com/acalatrava/openhaystack-firmware/blob/main/apps/openhaystack-alternative/iBeacon%20StepByStep.md), though a little outdated, since I used openhaystack back then - but I think it is adaptable. Maybe it helps you.

J0s3f, 2024-09-12:
I recently bought a few of this https://de.aliexpress.com/item/1005006269319766.html?spm=a2g0o.order_list.order_list_main.4.f29b5c5fI2G8SM&gatewayAdapt=glo2deu nrf51 devices and they work fine.

arlion-dev, 2024-11-18:
@J0s3f Have you found a datasheet to know the pin layout?

dschense, 2024-11-19:
yes pinout would be great. Maybe upload a image with some painting for the pins?

GunayAnach, 2025-01-06:
@dschense Check [this thread out](https://github.com/biemster/FindMy/issues/14#issuecomment-1367742980) for pinout of various ST17H66 based keychains

arlion-dev, 2025-01-08:
> @arlion-dev @dschense Check [this thread out](https://github.com/biemster/FindMy/issues/14#issuecomment-1367742980) for pinout of various ST17H66 based keychains for flashing with a TTL programmer 3v3 -> 3.3v+ GND -> GND TXD -> P9 RXD -> P10
> 
> nrf51 ibacon pinouts are usually [marked on the board silkscreen](https://ae-pic-a1.aliexpress-media.com/kf/HTB1nR8BKxGYBuNjy0Fnq6x5lpXap.jpg_960x960q75.jpg_.avif) with SWS and SWDI, which is how they match the ST-Link programmer

@GunayAnach I'm talk about this one :

[image]

GunayAnach, 2025-01-08:
@arlion-dev

That looks alot like a UART.  Which usually is read with USBtoTTL serial programmer

Ensure first one side is GND and other side is VCC

Pinouts should be these if its UART
[image]

RX and TX are crossed between the programmer and device

magnets110, 2025-02-24:
>  connect to stc flasher (below 1€) except power , 

What is stc flasher? just a ch340?

zen85, 2025-05-09:
> [@GunayAnach](https://github.com/GunayAnach) I'm talk about this one :
>
> [image]

can anybody here confirm or have an idea about the pinout on this?

because these:
https://de.aliexpress.com/item/1005006269319766.html?spm=a2g0o.cart.0.0.772d4ae4fVfJLI&mp=1&pdp_npi=5%40dis%21EUR%21EUR%204.50%21EUR%204.50%21%21EUR%204.50%21%21%21%402101eac917467905232244033ed923%2112000036543409236%21ct%21AT%212028404953%21%213%210&gatewayAdapt=glo2deu

in combination with something like this:
https://de.aliexpress.com/item/1005006269319766.html?spm=a2g0o.cart.0.0.772d4ae4fVfJLI&mp=1&pdp_npi=5%40dis%21EUR%21EUR%204.50%21EUR%204.50%21%21EUR%204.50%21%21%21%402101eac917467905232244033ed923%2112000036543409236%21ct%21AT%212028404953%21%213%210&gatewayAdapt=glo2deu

sounds like an extremly tempting solution to me... i could make a lot of them in bulk very fast without any soldering.

supaeasy, 2025-05-09:
You put the same link twice

zen85, 2025-05-09:
> You put the same link twice

fixed it... 

and of course this https://www.amazon.de/dp/B07YJPDDKQ?ref=fed_asin_title
would be necessary too - i am interessted in writing a guide in this setup because it would be beginnerfriendly if i get it too work