dchristl/macless-haystack #78: Updating Docker fails on DSM / Container Manager

supaeasy, 2024-04-23:
Hi there

I am getting this error when restarting Docker since release 2.2 and I don't know what to do. I would most likely start everything from anew but I am not sure if the problem is on my side. Any Idea what seems to be the issue here?

2024/04/23 09:12:16 | stdout | Aborting
2024/04/23 09:12:16 | stdout | Please commit your changes or stash them before you merge.
2024/04/23 09:12:16 | stdout | endpoint/config.py
2024/04/23 09:12:16 | stdout | error: Your local changes to the following files would be overwritten by merge:
2024/04/23 09:12:16 | stdout | Aborting
2024/04/23 09:12:16 | stdout | Please commit your changes or stash them before you merge.
2024/04/23 09:12:16 | stdout | endpoint/config.py
2024/04/23 09:12:16 | stdout | error: Your local changes to the following files would be overwritten by merge:
2024/04/23 09:12:16 | stdout | * branch            main       -> FETCH_HEAD
2024/04/23 09:12:16 | stdout | From https://github.com/dchristl/macless-haystack

I am running this on DSM7 in Container Manager and added it via [docker hub](https://registry.hub.docker.com/r/christld/macless-haystack/) if that helps.

dchristl, 2024-04-23:
The issue here is that you changed files inside the container. The easiest way to solve this is to remove the container and install it again.

supaeasy, 2024-04-23:
I dont think I changed files inside the container as I wouldnt know, how to.
I did a fresh start. Now I get these errors when I try to use the old auth.json and certificate.pem files and the container crashes.

2024/04/23 14:07:54 | stdout | raise NoSectionError(section) from None
2024/04/23 14:07:54 | stdout | File "/usr/local/lib/python3.12/configparser.py", line 1132, in _unify_values
2024/04/23 14:07:54 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:54 | stdout | d = self._unify_values(section, vars)
2024/04/23 14:07:54 | stdout | File "/usr/local/lib/python3.12/configparser.py", line 759, in get
2024/04/23 14:07:54 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:54 | stdout | value = config.get('Settings', 'binding_address')
2024/04/23 14:07:54 | stdout | File "/app/endpoint/config.py", line 36, in getBindingAddress
2024/04/23 14:07:54 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:54 | stdout | httpd = HTTPServer((config.getBindingAddress(), config.getPort()), Handler)
2024/04/23 14:07:54 | stdout | File "/app/endpoint/mh_endpoint.py", line 163, in <module>
2024/04/23 14:07:54 | stdout | Traceback (most recent call last):
2024/04/23 14:07:54 | stdout | * branch            main       -> FETCH_HEAD
2024/04/23 14:07:54 | stdout | From https://github.com/dchristl/macless-haystack
2024/04/23 14:07:14 | stdout | configparser.NoSectionError: No section: 'Settings'
2024/04/23 14:07:14 | stdout | raise NoSectionError(section) from None
2024/04/23 14:07:14 | stdout | File "/usr/local/lib/python3.12/configparser.py", line 1132, in _unify_values
2024/04/23 14:07:14 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:14 | stdout | d = self._unify_values(section, vars)
2024/04/23 14:07:14 | stdout | File "/usr/local/lib/python3.12/configparser.py", line 759, in get
2024/04/23 14:07:14 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:14 | stdout | value = config.get('Settings', 'binding_address')
2024/04/23 14:07:14 | stdout | File "/app/endpoint/config.py", line 36, in getBindingAddress
2024/04/23 14:07:14 | stdout | ^^^^^^^^^^^^^^^^^^^^^^^^^^
2024/04/23 14:07:14 | stdout | httpd = HTTPServer((config.getBindingAddress(), config.getPort()), Handler)
2024/04/23 14:07:14 | stdout | File "/app/endpoint/mh_endpoint.py", line 163, in <module>
2024/04/23 14:07:14 | stdout | Traceback (most recent call last):
2024/04/23 14:07:14 | stdout | * branch            main       -> FETCH_HEAD
2024/04/23 14:07:14 | stdout | From https://github.com/dchristl/macless-haystack

when I try it with an empty folder, I get to enter my credentials and then the same errors happen -> container crashes.

dchristl, 2024-04-23:
It seems your config.ini is corrupt. Can you delete this file and restart your container? It is in your Docker volume folder besides the certificate. pem

supaeasy, 2024-04-23:
Funny thing: There was no config.ini at all when all the hassle started... 
After trying out 2.1.0 without success, then 2.0.0 which finally "worked" (=didnt crash) It created a config.ini that I kept. I had to manually set anisette IP because it's name was not resolvable, though. I rebooted my NAS, tried again with 2.2.0 and this config.ini and it worked. A little strange but It seems to work now. Will keep an eye on this. It is weird because I don't remember editing config.ini when I initially set it up and I really dont see the reason why try No. 18 worked as I really don't think I did things differently. But well - closing this as it might have been a problem on my NAS.

closed 2024-04-23 by supaeasy