dchristl/macless-haystack #80: Cannot add / generate new Accessories when self-hosting the Webapp

ThomasW2005, 2024-04-25:
I self-hosted the website using express.js on my VPS and can also access the website, but when I want to add the JSON file or generate a new device, it does not work, and I get the error that `generateKey` is not defined:
[image]

dchristl, 2024-04-26:
Hello,

have you  read [the FAQ](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-host-my-own-web-frontend) about it and why you shouldn't host your own frontend? I also think that I will remove the frontend in one of the next releases. 

The provided zip file only works if it runs in the root folder of your server; otherwise, you have to adjust the base href in the index.html. That's usually the problem.

ThomasW2005, 2024-04-27:
Oh yeah sorry didn't read that, works now, Thanks!

closed 2024-04-27 by ThomasW2005