dchristl/macless-haystack #82: no positions with multi key files

dschense, 2024-04-29:
I'm not quite sure if I'm not doing something wrong.

With a single key, generated via generate_keys.py, everything seems to work as intended.
The firmware can be patched and the dongle can be flashed so that location data can also be retrieved.

But as soon as I generate multiple keys (no matter how many: 5, 20, 50, 100,..), I can flash the patched firmware to the NRF51822 dongle, but I never get a location.
I have tested this procedure several times. Only once did I succeed in patching the firmware with a keyfile containing 20 keys and was able to send location data after the flash.

Is there anything I have overlooked?

Translated with DeepL.com (free version)

1Richter, 2024-04-29:
what was the command, where you added multiple keys to the .py script?
Have you tried to change the generate_keys.py itself with changing `MAX_KEYS` to 20, and the default in the `-n` argument to 20 and then make the command as you would with 1 key? 
Maybe there is a missing blank line at the end of the keyfile? If this works, the keyfile parser woould't stop until blank line I guess.

dschense, 2024-05-02:
Its fixed for me.

did not see the `MAX_KEYS` value within the firmware. Keyfile with `50`, but firmware with `20` does not match.
my fault.

closed 2024-05-02 by dschense