dchristl/macless-haystack #83: Google FindMy

supaeasy, 2024-04-30:
Are you planning on integrating Googles FindMy Network into this, so to use the same trackers both in Apples and Googles network?

dchristl, 2024-05-01:
Hello, 
I would really like to integrate that, possibly even bilaterally, so that a device distributes data over both Apple and Android, and the app queries both servers and the app displays this transparently in one UI. This would provide optimal coverage. Unfortunately, I just don't have the time to get into that, and the Android network isn't that advanced yet.

supaeasy, 2024-05-01:
That would be awesome and truly best of both worlds: one tracker using both networks in a densely populated area would almost mean real time tracking. 

Hey, no pressure. I am happy that you are considering this. That is enough for me atm. Thanks a lot for your marvellous work!

dschense, 2024-05-03:
@supaeasy @dchristl 
https://github.com/biemster/FindMy/issues/51

maybe this is a start for everybody who wants to help.

dkgs2000, 2024-08-25:
https://developers.google.com/nearby/fast-pair/specifications/extensions/fmdn

Here is in-depth technical detail of the google's findMy network BLE protocol.

Just leaving this here.... :)