dchristl/macless-haystack #86: Account limit reached

MeierAlex, 2024-05-03:
HAllo,

I did setup macless-haystack like described and was able to log in with my (newly created apple ID - I do not own a apple device). After providing the Server with my ID, Password and a valid 2FA (server triggers SMS to my phone) I get the following error:

```
024-05-03 06:58:19,325 - DEBUG - Answer from icloud login
2024-05-03 06:58:19,325 - DEBUG - {'dsid': '21697078426', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'Account limit reached.'}}, 'status': 0}
2024-05-03 06:58:19,326 - ERROR - Invalid status: 1
2024-05-03 06:58:19,326 - ERROR - Error message: Account limit reached.
2024-05-03 06:58:19,327 - ERROR - Unable to proceed, program will be terminated.
```

What do I have to do to get it working ?

dschense, 2024-05-03:
https://github.com/dchristl/macless-haystack/blob/main/FAQ.md
Must be "Error with registration"

Most new IDs without any real registered Apple Device are not working.
just take a device from a friend, login, logout and use the ID again.

dchristl, 2024-05-03:
Your device seems to have too much Apple-Ids registered. There are several findings in the web about this topic, like [this](https://discussions.apple.com/thread/253006541?sortBy=best)

dchristl, 2024-05-03:
The docker container is also a device ;) Have you tried it several times with different accounts?

dchristl, 2024-05-03:
This is strange. Can you log in at the [Apple website](https://appleid.apple.com/)? Maybe you can create  a new Apple-ID and try again.

DrSchottky, 2024-05-07:
Also happening to me.
Apple ID created years ago but never associated an Apple device. Tried adding CC but didn't help. Login from browser works fine

dmr138, 2024-05-08:
I'm getting the same error here using a recently created apple ID

DrSchottky, 2024-05-08:
Solved it by linking a (real) Mac to the account

dmr138, 2024-05-08:
Thank you much, That did the trick. Were you perhaps able to sign out of/ remove that mac from the account later on?

DrSchottky, 2024-05-09:
Yes and it kept working. Didn't try to remove the fake device and redo the login tho.

bernikr, 2024-10-20:
I ran into the same issue with a fresh apple account. Since I currently don't have access to any apple devices I'm still looking for workarounds.

What partially worked for me, was signing into www.icloud.com with the account and accepting the terms of service. At least that changed the error message from `Error message: Account limit reached.` to

```
2024-10-20 10:16:25,475 - DEBUG - Answer from icloud login
2024-10-20 10:16:25,476 - DEBUG - {'dsid': '22073806685', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'A server problem is blocking Apple ID sign in. Try signing in later.'}}, 'status': 0}
2024-10-20 10:16:25,476 - ERROR - Invalid status: 1
2024-10-20 10:16:25,476 - ERROR - Error message: A server problem is blocking Apple ID sign in. Try signing in later.
2024-10-20 10:16:25,476 - ERROR - It seems your account score is not high enough. Log in to https://appleid.apple.com/ and add your credit card (nothing will be charged) or additional data to increase it.
2024-10-20 10:16:25,476 - ERROR - Unable to proceed, program will be terminated.
```

I've tried adding a credit card to the account, but this message still persists

leonardoviada, 2024-12-04:
> I ran into the same issue with a fresh apple account. Since I currently don't have access to any apple devices I'm still looking for workarounds.
>
> What partially worked for me, was signing into [...]

Same result here after logging into iCloud with a fresh account. Will try again by registering an Apple Device and post the progress

AidanHN, 2024-12-30:
I have the same issue as @bernikr. I have no Apple devices to register to the new ID but I get the account score error. Is there any solution?

iMartyn, 2025-01-14:
I also had this issue with an old apple account and could not get a new account to log in fully to enable 2fa. (https://www.reddit.com/r/applehelp/comments/17u6h66/apple_id_websites_continue_button_ignores_me/ shows this is very common)

This caused me with no apple devices to go down the rabbit hole of virtual machines.  I finally had success with https://github.com/sickcodes/Docker-OSX (which is osx in kvm orchestrated by docker, not dockerised osx, before you get excited).  Once I logged in on the virtual mac, my account unlocked.

Finding some easier way to unlock the accounts would be awesome though.

KUKHUA, 2025-01-15:
> I also had this issue with an old apple account and could not get a new account to log in fully to enable 2fa. [...]

Yep, you can use a virtual Mac to get it to work. I didn't have luck with docker-OSX, so I used https://oneclick-macos-simple-kvm.notaperson535.is-a.dev/ to get up a virtual machine. Also make sure you change your virtual Mac's serials **before signing into it**.

magnets110, 2025-02-05:
Does the apple account need to be logged in to a mac or can it just be an old iphone?

dchristl, 2025-02-06:
> https://dchristl.github.io/macless-haystack/

This can be any Apple device