dchristl/macless-haystack #87: Successfully enabled 2FA, but account is blocked | It seems your account score isn't high enough. What accounts have you successfully accessed, guys

Ramenaru, 2024-05-06:
</plist>
2024-05-06 04:20:46,235 - DEBUG - Answer from icloud login
2024-05-06 04:20:46,236 - DEBUG - {'dsid': '21623485671', 'delegates': {'com.apple.mobileme': {'status': 1, 'status-message': 'A server problem is blocking Apple ID sign in. Try signing in later.'}}, 'status': 0}
2024-05-06 04:20:46,236 - ERROR - Invalid status: 1
2024-05-06 04:20:46,236 - ERROR - Error message: A server problem is blocking Apple ID sign in. Try signing in later.
2024-05-06 04:20:46,237 - ERROR - It seems your account score is not high enough. Log in to https://appleid.apple.com/ and add your credit card (nothing will be charged) or additional data to increase it.
2024-05-06 04:20:46,237 - ERROR - Unable to proceed, program will be terminated.

dchristl, 2024-05-10:
Hello,

this is explained in the error message itself ( It seems your account score is not high enough. Log in to https://appleid.apple.com/ and add your credit card (nothing will be charged) or additional data to increase it.) and in the [FAQ](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#error-during-registration).

The problem is not every account works and only Apple knows why. It can also help to create a new one.

filip2cz, 2025-05-01:
I had same issue and it was solved when I had a chance to login to that account on some macbook, which was first time on that account. I believe that after this, Apple sees this account as more "trusted"