dchristl/macless-haystack #88: Map is jumping back from zoom level (App)

supaeasy, 2024-05-07:
Hi there 

whenever I move the map or zoom in on an area it takes between 3 and 5 seconds until map zoom level reverts back to default (=my position and the furthest accessory being the boundaries of the map). This is very annoying. Is there a way do disable this beaviour or could you make it so, that this only happens on refreshing position data?

Thanks a lot

dchristl, 2024-05-10:
Hello,

does this happen only after a fresh report fetching or always? Do you use the browser version? If the issue is after fetching and it is the browser version, the problem might be performance. After fetching only the latest report will be directly shown all other ones will be decrypted in the background, This takes a while and when finished the map will be redrawn (maybe some of the later decrypted location is the current location). So could you fetch fresh reports, wait one or two minutes and check if the problem still exists? If so, I will check if I can do some optimizations. Thank you

supaeasy, 2024-05-10:
Hello, no this problem is App-related and happens unrelated to position reports loading.
I tried the browser version (that I hardly ever use) and it reverts the map to default when I change tabs. But not while using it.

dchristl, 2024-05-24:
> no this problem is App-related and happens unrelated to position reports loading

Could you please take a video, I'm unable to reproduce this. The browser version reverts the map, this is OK, because it is not easy to differentiate between first visit and revisit, so the refresh/ reset is everytime

MeierAlex, 2024-05-27:
Same problem here - how can I provide you the video ? I do not weant to upload it here because it contains some personal informations.

dchristl, 2024-06-03:
Could you upload it on any hoster, Google Drive or Dropbox and send me a link to smsoip{at}gmail{dot}com. You can also zip it and decrypt it with a password. I will have a look when I have time.

ReLoia, 2026-04-05:
Hi! I this 'problem' is related to the method 
```dart
fitToContent(accessories, locationModel.here);
```
executed in 
map.dart -> Build()

this can be easily fixed by removing that method or adding a toggle in the settings to disable it if not needed like here:
```dart
      if (Settings.getValue<bool>(shouldAutoresizeOnLocationsKey,
          defaultValue: true)!) {
          fitToContent(accessories, locationModel.here);
      }
```

in my repo you can find my commits where I added some personal fixes to some things of the app I didn't like and you may build from it or copy my code (it's not a great code):
https://github.com/ReLoia/macless-haystack/commit/42abb3c633c10f87f4b24ffd7d83b8ad5d59ac99

tobiasbayer, 2026-04-13:
In my personal opinion switching the boolean on map interaction would be the ideal way of fixing this. But I think the settings switch you introduced would also be a good enough solution. 
Note that I am not the maintainer of the repo, so I can only speak for myself and my own preferences.