dchristl/macless-haystack #92: Expose macless-haystack to the internet

MeierAlex, 2024-05-21:
Is there anything you should be aware of apart from activating user name and password protection? Is there any experience with this? 

Best regardss.
Alex

Systm21, 2024-05-21:
You should use a Reverseproxy or an Cloudflared Argotunnel to add another layer of Security.

dchristl, 2024-05-24:
Hello,

you should use at least SSL encryption, i.e. with [Let's encrypt](https://letsencrypt.org)

Regards,
Danny

MeierAlex, 2024-05-25:
I tired exposing the interface to the internet by forwarding port 6176, setting up SSL (like described in the FAQ) and enabeling authntification. Now I can not reach the server from the internet. When I try to reach "my-server-name:6176" I just get a timeout. An I doing something wrong ?

dchristl, 2024-05-26:
Have you used https instead of http? Can you reach the url in your browser? You should see a website with content "Nothing to see here"

MeierAlex, 2024-05-26:
I used https and tried to reach the website resulting in a timeout.

dchristl, 2024-05-26:
What does the [log](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-see-the-logs) say? Is it reachable in general on your server (with https://localhost:6176)?

MeierAlex, 2024-05-26:
Locally is everything reachable. I will post the log later.

MeierAlex, 2024-05-27:
Problem solved - I had a problem with setting up subdomains. Direct port forwarding did work when I restarted the macless haystack server after setting up the forwarding in the router.

closed 2024-05-27 by MeierAlex