dchristl/macless-haystack #95: The issue with the total number of tags

haohuoq, 2024-05-22:
Is the public-private key generated by the generate.keys Python script bound to the AppleID registered by Docker in theory? In other words, is there a correlation between the generated public and private key pairs and the Docker's AppleID?
I don't think this public-private key pair has anything to do with the AppleID, which is just a request for the identity of the Apple server.
According to the information, an AppleID can be bound to a maximum of 16 AirTags. Is there no limit to the number of Tags I can create? Can I create more than 16 AirTags?

haohuoq, 2024-05-22:
The cfg file of openocd needs to be changed from nrf51 to 52, and the compilation command also needs to be modified
I am using nrf51 mass_erase instead of nrf52 mass_erase because nrf52 mass_erase reported an error. My command is as follows
[image]

”openocd -f openocd.cfg -c "init; halt; nrf51 mass_erase; program nrf52_firmware.bin; reset; exit" 
Then I successfully burned the nrf52 platform and viewed the location information

dchristl, 2024-05-24:
Hello,

no it's not bound to any account and there is no correlation between the container and the tag. The container just downloads the encrypted locations. In theory you can download encrypted reports of any device. For the decryption you need the private key. Therefore you have no such limitation with macless-haystack.

Sorry, about the issue with the nrf52, I sadly don't have the time to support this device anymore. Maybe [@jrperson ](https://github.com/jrperson) can help you out.