dchristl/macless-haystack #96: cant get it working, how to debug?

johnnyrun, 2024-05-22:
Hi all!
I have a esp32-wroom-32 device, detected as "ESP32-D0WDQ6 (revision v1.0)" during flash.
All steps are fine but I cant get it working

in serial I get:
```
E (952) macless_haystack: Found 50 keys
I (952) macless_haystack: Loading key with index 45 at address 1261
I (962) macless_haystack: using key with start 90 09
I (962) macless_haystack: using device address: d0 09 f3 d1 f7 9d
I (972) macless_haystack: Sending beacon (with key index 45)
I (992) macless_haystack: advertising has started.
I (1082) macless_haystack: stop adv successfully
I (1182) macless_haystack: Going to sleep
```

so it seems I flashed correctly.
macless is running fine but returns always empty array:

```
{"results":[],"statusCode":"200"}
````

I have an Iphone on the same table, with default settings.
How may I debug that?

I'm  tring with android "apple find tracker" with no luck..
I would like to exclude an hardware problem..
Thanks

giannicarabelli, 2024-05-22:
I add some details:

downsized to 1 key, installed BLE scanner (a mobile app that listen BLE beacon) and I found the device address, as seen in log. 
So it seems it's not an hardware problem.. but still:

```
2024-05-22 17:16:20,180 - DEBUG - Getting with post: b'{"ids":["kkUrbhjd8dit2Mh/2DtFjaBjAiuydtWGIDyWic+x7PM="],"days":6}'
2024-05-22 17:16:20,180 - DEBUG - Querying for 6 days
2024-05-22 17:16:20,483 - DEBUG - Return from fetch service:
2024-05-22 17:16:20,484 - DEBUG - {"results":[],"statusCode":"200"}
```

johnnyrun, 2024-05-22:
I add some details:

downsized to 1 key, installed BLE scanner (a mobile app that listen BLE beacon) and I found the device address, as seen in log.
So it seems it's not an hardware problem.. but still:

```
2024-05-22 17:16:20,180 - DEBUG - Getting with post: b'{"ids":["kkUrbhjd8dit2Mh/2DtFjaBjAiuydtWGIDyWic+x7PM="],"days":6}'
2024-05-22 17:16:20,180 - DEBUG - Querying for 6 days
2024-05-22 17:16:20,483 - DEBUG - Return from fetch service:
2024-05-22 17:16:20,484 - DEBUG - {"results":[],"statusCode":"200"}
```

dchristl, 2024-05-24:
Hello,

it seems everything is working fine. Both outputs look great. You can either go with your tag to a crowded area (Apple had several IPhone-releases, that doesn't submit any reports) or check if you have imported the same json that is generated during the key-generation process.