dchristl/macless-haystack #98: Firmware builds for ESP32-C3

yanxke, 2024-05-25:
For some reason I can't get the build environment set up.  Is it using PlatformIO or the native ESP IDF extension?  Is it possible to provide the prebuilt firmware for the ESP32-C3?

dchristl, 2024-05-26:
It is using PlatfomIO with ESP IDF. You only need to open the platfomio.ini-file with PaltformIO and you should be fine. The firmware is the same for all ESP32 flavors, so can download the binary in the releases of this project.

yanxke, 2024-05-26:
The C3s are on the RISC-V architecture, as opposed to Xtensa for the original ESP32s, and so the firmware needs to be recompiled in order to work.

yanxke, 2024-05-26:
Even running the build on the default configuration fails.  I suspect it's a mismatch in the espressif32 platform version.
```

 *  Executing task: platformio run 

Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50201.240515 (5.2.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp-elf @ 13.2.0+20230928
Reading CMake configuration...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp32dev/src/openhaystack_main.c.o
Compiling .pio/build/esp32dev/bootloader_support/src/bootloader_mem.c.o
Compiling .pio/build/esp32dev/bootloader_support/src/bootloader_random.c.o
Compiling .pio/build/esp32dev/bootloader_support/src/bootloader_efuse.c.o
src/openhaystack_main.c:148:1: error: unknown type name 'uint'; did you mean 'int'?
  148 | uint get_key_count()
      | ^~~~
      | int
src/openhaystack_main.c: In function 'app_main':
src/openhaystack_main.c:175:5: error: unknown type name 'uint'; did you mean 'int'?
  175 |     uint key_count = get_key_count();
      |     ^~~~
      |     int
src/openhaystack_main.c:176:5: error: unknown type name 'uint'; did you mean 'int'?
  176 |     uint key_index = (esp_random() % key_count);
      |     ^~~~
      |     int
Compiling .pio/build/esp32dev/bootloader_support/src/flash_encrypt.c.o
Compiling .pio/build/esp32dev/bootloader_support/src/secure_boot.c.o
Compiling .pio/build/esp32dev/bootloader_support/src/bootloader_random_esp32.c.o
*** [.pio/build/esp32dev/src/openhaystack_main.c.o] Error 1
=========================================================================== [FAILED] Took 1.95 seconds ===========================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

yanxke, 2024-05-27:
Eventually got it to compile and broadcast beacons.  See https://github.com/dchristl/macless-haystack/pull/99

thirstyone, 2024-06-12:
so, finally, it works with c3! nice! 
have you assessed the real power consumption on those newish chips (compared to the classical nrf51822?)

yanxke, 2024-06-12:
The ESPs are orders of magnitude higher in power consumption than the NRFs.  Wouldn't want to battery power these.

solarkraft, 2024-08-28:
Is this a natural law or maybe just due to a lack of optimization? Anecdotally I heard that the new ESPs are quite power efficient (compared to the older ones) and I may even have seen figures that put them in a similar ball park as the most power efficient chips (when fully optimized).