dchristl/macless-haystack #100: Feature-Request: Battery-Information for NRF51xx

MeierAlex, 2024-05-28:
Hallo,

Is there a way to integrate a battery status in a way that it can be displayed in the (web-)app ? It looks like there is alredy something going on in an other project and I think this information would be very benificial to be able to distinguish between a tag with no network vs a tag with dead battery.

Here is a link to the other project (I hope it is ok to post it here):
https://github.com/dakhnod/FakeTag/blob/02755369eb18e34bb0cfb76a2dd165136ee35aa3/main.c#L33-L38

Thank you,
Alex

dchristl, 2024-06-03:
Hello,

this is not possible, because the project supports various tags and most of them have this feature.

Kind Regards,
Danny

MeierAlex, 2024-06-03:
Hallo,

so is this feature already implemented in the Backend/Webinterface and just needs to be enabled on the tag hardware or do you mean that this is not supported beause some tags support this "natively" ? If so can you give me an example of such a tag ? I would be very happy to have this feature for my tags because otherwise it ill be very hard to identify a tag with a low battery. 

Do you have another solution in mind for the battery issue? Basically, firmware adaptations and self-compiling are no problem for me.

cuonghm90, 2024-06-09:
> Ê,
> 
> vậy tính năng này đã được triển khai trong Phần cuối/Giao diện web và chỉ cần được bật trên phần cứng thẻ hay ý bạn là tính năng này không được hỗ trợ vì một số thẻ hỗ trợ "nguyên bản" này? Nếu vậy bạn có thể cho tôi một ví dụ về thẻ như vậy không? Tôi sẽ rất vui khi có tính năng này cho thẻ của mình vì nếu không sẽ rất khó xác định thẻ có pin yếu.
> 
> Bạn có giải pháp nào khác cho vấn đề pin không? Về cơ bản, việc điều chỉnh phần sụn và tự biên dịch không có vấn đề gì đối với tôi.

This is the feature I want this time. Can you dev it ?

cuonghm90, 2024-06-09:
This is the feature I want this time. Can you dev it ? 

> Hallo,
>
> so is this feature already implemented in the Backend/Webinterface and just needs to be enabled on the tag hardware or do you mean that this is not supported beause some tags support this [...]

dchristl, 2024-06-10:
No this feature is not implemented, because a lot of Tags are supported and most (I think all, except NRF5x) doesn't have this feature. Most users uses 1$ devices from Aliexpress without this feature. It seems nice to add such a feature, but only for a couple of users and I sadly really don't have time for that currently.

> Do you have another solution in mind for the battery issue? Basically, firmware adaptations and self-compiling are no problem for me.

Not really, you have to extend the firmware to send the battery informations and also to read it out in the flutter app. This seems doable, but there have to be a little bit research.

MeierAlex, 2024-06-11:
> Most users uses 1$ devices from Aliexpress without this feature
Can you tell about which devices you are talking here ? I would be very interested ;)

I think the changes in ther NRF Firmware would not be that much - if you ever think of implementing this feature I can offer to do the changes for the NRF51 (looking at the exiasting code it seems like "piggybacking" some additional data to the location reports is not a problem at all). So if you decide to do the changes on the flutter app (which I can not do), just drop me a message for the NRF implementation ;)

cuonghm90, 2024-06-13:
> I think the changes in ther NRF Firmware would not be that much - if you ever think of implementing this feature I can offer to do the changes for the NRF51 (looking at the exiasting code it seems [...]

Because I need this feature, so I will implement soon.

dchristl, 2024-06-17:
Like I mentioned before, I will not have time for that in the near future. If you would like to add this, I would be happy to look at pull requests