dchristl/macless-haystack #101: Can I extend location history more 7 days ?

cuonghm90, 2024-05-30:
Can I extend location history more 7 days ?
I need tracking in 15~30 days.

dchristl, 2024-06-03:
This is currently not possible. Btw. the history is not provided by Apple, but by Macless Haystack. The app stores previously retrieved locations and keeps them available. So if you don't query for 2 days, it is possible that data in the history will be missing. This becomes more likely as the time period increases.

utya, 2024-06-06:
How long is the period for which Apple provides the information?

nk-gdsc, 2024-06-07:
7days
[image]
https://dl.acm.org/doi/abs/10.1145/3448300.3468251

cuonghm90, 2024-06-09:
> This is currently not possible. Btw. the history is not provided by Apple, but by Macless Haystack. The app stores previously retrieved locations and keeps them available. So if you don't query for [...]

Can I save respone to db on server ? 
It will extend any day I want.
When client send request to server, server will fetch data from Apple and update to db, after that server respone data got from db to client.

dchristl, 2024-06-10:
You can have a look at [Biemster's project](https://github.com/biemster/FindMy). It is without an UI, but location data will be saved to a db. And the two projects are compatible.