dchristl/macless-haystack #104: Front-end won't display points on map even though it claims locations have been fetched

cwiokpl, 2024-06-09:
Hello,

I installed macless-haystack on a fresh Debian 12 VM all according to the guide. I have a tag that is confirmed to work and I am able to extract locations using OpenHaystack MacOS app. When I try to use Macless-Haystack I get info that locations have been fetched - I see the request and response in Dev Tools (btw how can I extract coords from that), but it won't show any points on map. Two screenshots:

[image]
[image]

Any idea what might be causing this? All help appreciated.

dchristl, 2024-06-10:
Hello,

the map normally adjusts itself after fetching to the latest coordinates. Also, if you click on the device in the list, the map will center automatically. 
The error of the browser console is known (not fixed yet by the maintainer), but should not harm anything. You can't see the coordinates directly by the requests. Because everything is encrypted and can only be decrypted by your private key. 
It is hard to help you here without the reports/location data, because seems working fine. The only known reason, why no location is shown, if all reports are invalid (strange lan/lot values)

cwiokpl, 2024-06-20:
@dchristl  Thank you for the reply. As I mentioned it was installed on a fresh VM, so I am too quite surprised it is not working. What other additional logs would I need to supply for us to debug this? Thank you for help.

dchristl, 2024-06-22:
Have you already jumped to the place on the map where the devices should be?  
The only thing can that can help here is your accesory.json-file to decrypt your reports and debug inside the code. Alternatively, you can try this yourself in the Flutter app.