dchristl/macless-haystack #105: Self hosted webapp

MeierAlex, 2024-06-11:
Hallo,

I want to host the webapp on one of my servers. I did setup Apache2 and copied the webappfiles from the last release to /var/www/. I can open the webinterface but I can not import devices - if I import them from a .json file simply nothing happens. The same applies when i try to pick up data from the backend. i think there is something missing on my setup. i am running a server with debian 12 and apache2 installed - nothing more at the moment. Can you give me a hint, please ?

Best regards,
Alex

dchristl, 2024-06-17:
Hello,

have you read the [FAQ](https://github.com/dchristl/macless-haystack/blob/main/FAQ.md#how-can-i-host-my-own-web-frontend) why this isn't needed and shouldn't be done? I will no longer offer the frontend for download in one of the next versions and will also no longer support it

Best Regards,
Danny

thirstyone, 2024-07-20:
> the next versions and will also no longer support it
that's a pity. sometimes it's handy to get the map from another provider. 
from my understanding, all the locations data is served by the macless-haystack. however, when opening localhost:6176, it says 'nothing to see here'. is there any documentation about API of this server? 

and another question regarding the mobile client - any precompiled apks, by any chance?

dchristl, 2024-07-29:
> that's a pity. sometimes it's handy to get the map from another provider.

Sorry, I don't quite understand what you're trying to say. If you want to have the map from another provider, then you should recompile the Flutter application anyway, because there is currently no setting for it. 

> from my understanding, all the locations data is served by the macless-haystack. however, when opening localhost:6176, it says 'nothing to see here'. is there any documentation about API of this server?

The server part receives the public keys and sends them to Apple to get the reports. These encrypted reports are sent 1:1 to the client. This happens via POST when you call the URL in the browser (GET), then you get the output mentioned. There is no documentation, there is also only 1 interface. You can check out the [source](https://github.com/dchristl/macless-haystack/blob/main/endpoint/mh_endpoint.py) L68ff, if you are interested. The endpoint log also shows what is sent and received.

> and another question regarding the mobile client - any precompiled apks, by any chance?

Check out the [releases](https://github.com/dchristl/macless-haystack/releases)