dchristl/macless-haystack #106: how do I get the json file based on my array of advertisement keys?

thirstyone, 2024-06-17:
It is only apple that knows what is going on there, but today I made the third attempt (the third ISP) and this time successfully passed 3FA. I've got a set of working devices from my previous setup, but I don't know how to obtain a json file. any clue?

dchristl, 2024-06-22:
What do you mean by the JSON file? The file for import? It is generated during the generate_keys process.

thirstyone, 2024-07-26:
yep. Not needed anymore)
The reason I asked was your github frontend (or how do I call it) displayed a blank page in my safari (but I am on a non-metal machine, with OCLP version of Ventura). In Vivaldi it displayed just the icon. Interestingly, when in private mode, Vivaldi worked just fine. The thing is each time I wanted to find my device, I had to first open my initial openhaystack app to copy the private key, and then add a new device (this data is lost each time I close the browser window). 

I ended up with Chrome, which seems to work just fine. The only minor issue is I didn't find out how to zoom out the map. 

For 'tinkering' and educational purposes - trying to understand, 1. is there a way to request just the latest location, rather than all locations over the last day. 2. how do I decipher that json file (the response from macless-haystack on the localhost) into lat-long values.

dchristl, 2024-07-29:
> 1. is there a way to request just the latest location, rather than all locations over the last day.

This works with restrictions. You can set the time period in the settings. Then only the last report is retrieved. Unfortunately, this is not always accurate, as this is the last report that was sent to Apple and not the last location. These are usually identical, but it can happen that an older location was transferred later. 

> 2. how do I decipher that json file (the response from macless-haystack on the localhost) into lat-long values.

You can check out [biemster's project](https://github.com/biemster/FindMy) (especially [request_reports.py](https://github.com/biemster/FindMy/blob/main/request_reports.py)) for decrypting the reports from Apple (the response from macless haystack is identical)